KCL-Planning/ROSPlan #164: Bug in dispatcher

Samcheok, 2018-12-11:
Hello,

We found a bug in Dispatcher. 

Two combinations of versions are tested 
1 ) Errors with any domains

    - ROSPlan version :  https://github.com/KCL-Planning/ROSPlan/tree/7e2efc62fb6650b9df7de5f580811225016ed502
    - Esterel plan parser version : all Floyd-Warshall  
    - domain & problem : MapAnalyzer, P5-5-3-0-1

  Bug while dispatching, saying "Preconditions not achieved"

[image]

Domain, Problem, and Plan files are attached.
[attached_file.zip](https://github.com/KCL-Planning/ROSPlan/files/2666527/attached_file.zip)

2) Errors varies with domains
-  ROSPlan kinetic version : https://github.com/KCL-Planning/ROSPlan/tree/kinetic  

Any comments would be appreciated.

-- Kyeong Seok

oscar-lima, 2018-12-12:
Hi, I looked into your issue and have some comments. Basically the dispatcher is checking the KB to check if grounded action preconditions are met before dispatching the action. From your error there are many preconditions which are not satisfied and therefore the complaint from the dispatcher. I am trying to replicate your error but currently ROSPlan action simulator only supports fixed time durations, however I am extending the simulator interface to support it. Only after that I will be able to replicate your problem. An easier way would be if you could share your simulator code... greetings. Oscar

oscar-lima, 2018-12-19:
[image]

oscar-lima, 2018-12-19:
[image]

m312z, 2019-01-08:
Hello both,

For this plan (vehicle_start, build_road) to be dispatched correctly the precondition of the build_road action needs to be observed at 320.001, and before 320.002. This time is too short for the turnaround of the ROS service.

It is for to avoid tight coordination problems like this that PDDL enforces the epsilon separation between action. In this case the epsilon separation is too small, only 0.001 seconds.

If the planner supports it, then it would be a good idea to increase this value. Then the generated plan should be dispatched normally. 

From the plan validator README:

> VAL has many command line options, but the most important first few are:
> 
> validate -t -v <planfile....>
> 
> Multiple plan files can be handled together. The -t flag allows the value of epsilon to be set. The default value is 0.01, but 0.001 is a good value to use for most planners. Actions separated by epsilon or less are treated as simultaneous by VAL. -v is the verbose flag.

While 0.001 is what a lot of planners use, it is rather a small duration in robotics.

Michael