KCL-Planning/ROSPlan #168: Missing dependency in  ROSPlan/rosplan_dispatch_msgs/package.xml

cjmahieu, 2019-01-03:
Hi

There seems to be a missing dependency in ROSPlan/rosplan_dispatch_msgs/package.xml. Specifically 
'<build_depend>actionlib_msgs</build_depend>' and '<run_depend>actionlib_msgs</run_depend>'.

I am interfacing with rosplan using rosjava which means that I use genjava_message_artifacts to generate a jar containing the rosplan_dispatch_msgs messages. However this fails since the dependency is missing. Adding it has solves the issue.

Kind regards
Christof

cjmahieu, 2019-01-04:
I embarrassingly have to admit that I do not know how to do this.
From what I read online I need to be able to create a new branch first and push it to this repository before I can make the pull request but I do not seem to have access rights to do so.

I have made branch locally and commit the change there but I do not know how to proceed further.

Apologies
Christof

m312z, 2019-01-04:
Not embarrassing at all! Since you do not have write access to the repository, there are a few steps:
1. Create your own fork of the repository using the button on this page (a copy you have write access to).
2. Commit and push your change to your fork.
3, Open a pull request (using the buttons on the webpage).

Since you have an existing clone of the repo, then to push your changes to your fork you will first have to add it as a new remote. You can see how to do that here:
https://help.github.com/articles/adding-a-remote/

I hope this is fun,
Michael

closed 2019-06-25 by m312z