KCL-Planning/ROSPlan #172: Created a New client node for Problem_generator_server, is not generating the problem.

NagaDinesh1194, 2019-01-29:
Hey Guys,

I am trying to create a new client node for the problem_generation_server.
I am able to create a node and run it but it doesnt make a service call to the rosplan_generation_server.

I have placed this node file "nags_rosplan_client"  in 
"/home/nags/ROSPlan/src/rosplan/rosplan_planning_system/src/ProblemGeneration". 
And entered the details of the same in the 'Cmakelists.txt ' present in the following path, "/home/nags/ROSPlan/src/rosplan/rosplan_planning_system".

Mentioned the following lines in Cmakeslists.txt:
           add_executable(nags_rosplan_client src/ProblemGeneration/nags_rosplan_client.cpp)
           add_dependencies(nags_rosplan_client ${catkin_EXPORTED_TARGETS})
           target_link_libraries(nags_rosplan_client ${catkin_LIBRARIES})

Following is my client code:

 `#include "ros/ros.h"
//#include "rosplan_planning_system/ProblemGeneration/ProblemInterface.h"
#include "std_srvs/Empty.h"

#include <cstdlib>

int main(int argc, char **argv)
{
  ros::init(argc, argv, "nags_rosplan_client");
 //if (argc != 3)
 //{
 //ROS_INFO("Making service calls to problem_generation_server");
 //  return 1;
 //}

  ros::NodeHandle n;
  ros::ServiceClient client = n.serviceClient<std_srvs::Empty>("problem_generation_server");
  std_srvs::Empty srv;
  //srv.request.a = atoll();
  //srv.request.b = atoll(argv[2]);
  if (client.call(srv))
  {
    ROS_INFO("Nags_rosplan_client Responsed ");
  }
  else
  {
    ROS_ERROR("Failed to call service rosplan_client");
    return 1;
  }

  return 0;
}`

And then i use 'catkin build' in ROSPlan directory.

- Build gives me no errors.
- The node is created and i am able to use 'rosrun' to run the node.
The issue is don't see the problem being generated.
Please do tell me what am i doing wrong.

Thanks in advance.

m312z, 2019-01-29:
Hello Naga,

You don't mention what output is printed:
  ROS_INFO("Nags_rosplan_client Responsed ");
  ROS_ERROR("Failed to call service rosplan_client");

I'm guessing that it is the second one, and that the reason is the topic of the service is not correct. Currently you are calling the topic: "problem_generation_server". This looks like the name of the node, and not the name of its service.

Can you launch everything and then use the command:
> rosservice list
This will list all the current services. You can then use:
> rosservice info [service]
To get more information on a service, i.e., the message type, server node, etc.

Michael

NagaDinesh1194, 2019-01-29:
Hi Michael,

I did the "rosservice list" search and got the info about the same, prior to raising this issue. 
Node name is "/rosplan_problem_interface" and it offers two services and one of which is "problem_generation_server".

Yes i am not printing anything for the output. But i hope that should be fine because the output 
`[ INFO] [1548761133.187964962]: KCL: (/rosplan_problem_interface) (problem.pddl) Generating problem file.
[ INFO] [1548761133.223417017]: KCL: (/rosplan_problem_interface) (problem.pddl) The problem was generated.`
is expected to be displayed by ROS_INFO(output printed by Problem interface node when the service is called). Client node doesn't seem to be making the service call.

Naga

m312z, 2019-01-29:
Hello Naga,

The service topic is not prepended by the node name? The parameter to the service client constructor should include the whole line from the rosservice list. E.g., I would expect to create the client with:
ros::ServiceClient client = n.serviceClient<std_srvs::Empty>("/rosplan_problem_interface/problem_generation_server")

Followed by a waitForService call, to ensure that the service exists.

Best regards,
Michael