KCL-Planning/ROSPlan #176: Help getting POPF planner to spit out a simple plan

keerthanamanivannan, 2019-03-08:
### Description

I have a simple domain and problem PDDL files, but I get `Plan was unsolvable.` output from the planner.

### My environment
* ROS Distro: Kinetic
* OS Version: Ubuntu 16.04
* Source build
* Branch: master
* Git commit: e79d62a624f68429c5e4955fd64068d61c17905c

### Steps to reproduce
I cloned ROSPlan and followed few of the tutorials. 
I added my own domain and problem PDDL files in `rosplan_demos/common/aq/`

### Domain file
[image]

### Problem file
[image]

My launch file:
<?xml version="1.0"?>
<launch>

	
	<arg name="domain_path"	value="$(find rosplan_demos)/common/aq/domain.pddl" />
	<arg name="problem_path"	value="$(find rosplan_demos)/common/aq/problem.pddl" />

	
	<node name="rosplan_knowledge_base" pkg="rosplan_knowledge_base" type="knowledgeBase" respawn="false" output="screen">
		<param name="domain_path" value="$(arg domain_path)" />
		<param name="problem_path" value="$(arg problem_path)" />
		<param name="database_path" value="$(find rosplan_knowledge_base)/common/mongoDB/" />
		
		<param name="use_unknowns" value="false" />
	</node>

	
	<include file="$(find rosplan_planning_system)/launch/includes/problem_interface.launch">
		<arg name="knowledge_base"   value="rosplan_knowledge_base" />
		<arg name="domain_path"      value="$(arg domain_path)" />
		<arg name="problem_path"     value="$(find rosplan_demos)/common/problem.pddl" />
		<arg name="problem_topic"    value="problem_instance" />
	</include>

	
	<include file="$(find rosplan_planning_system)/launch/includes/planner_interface.launch">
		<arg name="use_problem_topic"    value="true" />
		<arg name="problem_topic"        value="/rosplan_problem_interface/problem_instance" />
		<arg name="planner_topic"        value="planner_output" />
		<arg name="domain_path"          value="$(arg domain_path)" />
		<arg name="problem_path"         value="$(find rosplan_demos)/common/aq/problem.pddl" />
		<arg name="data_path"            value="$(find rosplan_demos)/common/aq/" />
		<arg name="planner_command"      value="timeout 10 $(find rosplan_planning_system)/common/bin/popf DOMAIN PROBLEM" />
	</include>

	
	<include file="$(find mongodb_store)/launch/mongodb_store.launch">
		<arg name="db_path" value="$(find rosplan_knowledge_base)/common/mongoDB/"/>
	</include>

</launch>

### Expected behaviour
I launch this launch file, and call these two services:
1. `rosservice call /rosplan_problem_interface/problem_generation_server`
2. `rosservice call /rosplan_planner_interface/planning_server`

I should get a plan.pddl with the action "Stow" or something.

### Actual behaviour
Instead, I get this:

> [ INFO] [1552073160.804842525]: KCL: (/rosplan_planner_interface) (problem.pddl) Planning complete
> [ INFO] [1552073160.804941163]: KCL: (/rosplan_planner_interface) (problem.pddl) Plan was unsolvable.

### Backtrace or Console output
```process[replicator_node-8]: started with pid [16081]
[ INFO] [1552073067.156950095]: KCL: (/rosplan_problem_interface) Ready to receive
[ INFO] [1552073067.208967971]: KCL: (/rosplan_knowledge_base) Parsing domain
[ INFO] [1552073067.209099173]: KCL: (/rosplan_knowledge_base) Parsing domain: /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/domain.pddl.
[ INFO] [1552073067.209486743]: KCL: (/rosplan_knowledge_base) Parsing initial state
[ INFO] [1552073067.209554154]: KCL: (/rosplan_knowledge_base) Parsing Problem File: /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/problem.pddl.
[ INFO] [1552073067.209735903]: KCL: (/rosplan_knowledge_base) Waiting for MongoDB
[ INFO] [1552073067.211048186]: waitForService: Service [/message_store/delete] has not been advertised, waiting...
[ INFO] [1552073067.230798705]: KCL: (/rosplan_planner_interface) Ready to receive
[ INFO] [1552073068.066033253]: waitForService: Service [/message_store/delete] is now available.
MongoDB shell version: 2.6.10
connecting to: message_store
2019-03-08T13:24:28.123-0600 remove needs a query at src/mongo/shell/collection.js:299
[ INFO] [1552073068.125870943]: KCL: (/rosplan_knowledge_base) Ready to receive
[ INFO] [1552073072.126034682]: KCL: (/rosplan_problem_interface) (problem.pddl) Generating problem file.
[ INFO] [1552073072.141167715]: KCL: (/rosplan_problem_interface) (problem.pddl) The problem was generated.
[ INFO] [1552073072.141485859]: KCL: (/rosplan_planner_interface) Problem received.
[ INFO] [1552073160.799022064]: KCL: (/rosplan_planner_interface) (problem.pddl) Writing problem to file.
[ INFO] [1552073160.799326552]: KCL: (/rosplan_planner_interface) (problem.pddl) Running: timeout 10 /home/kmanivannan/a_ws/install/share/rosplan_planning_system/common/bin/popf /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/domain.pddl /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/problem.pddl > /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/plan.pddl
[ INFO] [1552073160.804842525]: KCL: (/rosplan_planner_interface) (problem.pddl) Planning complete
[ INFO] [1552073160.804941163]: KCL: (/rosplan_planner_interface) (problem.pddl) Plan was unsolvable.```

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

DerekLong101, 2019-03-10:
Hi

POPF does not handle negative preconditions (at start (not (stowed ?a))). Try using a paired predicate, unstowed:

(at start (unstowed ?a))

:effect (and (at end (stowed ?a)) (at start (not (unstowed ?a))))

etc. Note the idiom in which you delete the unstowed at start and only add the stowed at end - in the middle of the action, the arm is neither stowed or unstowed. This is actually a good thing - because it stops any plans in which the arm is exploited while in the middle of stowing or unstowing.

Cheers

Derek

[d-pddl]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F15019704%2F54051335-31a8ca00-41a7-11e9-9a3c-e4e560cd3d75.png&data=01%7C01%7Cderek.long%40kcl.ac.uk%7Cb269d775e96c4c8fcf6208d6a3fd751e%7C8370cf1416f34c16b83c724071654356%7C0&sdata=yhgRjMvX8MJf5blwT1d4Rz2LhJSJP2Rde5Z%2FRK57iHw%3D&reserved=0>

[p-pddl]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F15019704%2F54051341-379eab00-41a7-11e9-9bb1-59c759d04046.png&data=01%7C01%7Cderek.long%40kcl.ac.uk%7Cb269d775e96c4c8fcf6208d6a3fd751e%7C8370cf1416f34c16b83c724071654356%7C0&sdata=znrt9fuPQ%2FE5X6giTMYG5KrDF9S4KiQtHIw6Jbk%2Fo9A%3D&reserved=0>

keerthanamanivannan, 2019-03-13:
Hi Derek, 

Thank you for the response. What you explained makes a lot of sense. 

However, when I changed the domain file to:
[image]

and problem file was the same:
[image]

I still get an error:
```
[ INFO] [1552505729.284645504, 739.787000000]: KCL: (/rosplan_planner_interface) (problem.pddl) Running: timeout 10 /home/kmanivannan/a_ws/install/share/rosplan_planning_system/common/bin/popf /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/domain.pddl /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/problem.pddl > /home/kmanivannan/a_ws/install/share/rosplan_demos/common/aq/plan.pddl
Critical Errors Encountered in Domain/Problem File
--------------------------------------------------

Due to critical errors in the supplied domain/problem file, the planner
has to terminate.  The errors encountered are as follows:
[ INFO] [1552505729.295531608, 739.789000000]: KCL: (/rosplan_planner_interface) (problem.pddl) Planning complete
[ INFO] [1552505729.295634449, 739.789000000]: KCL: (/rosplan_planner_interface) (problem.pddl) Plan was unsolvable.
```

I'm getting some syntax errors in the domain file:
```./tan ../rosplan/rosplan_demos/common/aq/domain.pddl 
File: ../rosplan/rosplan_demos/common/aq/domain.pddl

Errors: 2, warnings: 0
../rosplan/rosplan_demos/common/aq/domain.pddl: line: 24: Error: Syntax error in durative-action declaration.
../rosplan/rosplan_demos/common/aq/domain.pddl: line: 24: Error: Unreadable structure```

Do you see any obvious issues here that I'm missing?

DerekLong101, 2019-03-13:
If that is your pddl file, then one error is the missing bracket at end of line 23. Try running popf outside rosplan first, to debug your domain.

Cheers

Derek

[Screenshot from 2019-03-13 14-52-12]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F15019704%2F54310114-9d23da80-459f-11e9-88cb-351ec0022596.png&data=01%7C01%7Cderek.long%40kcl.ac.uk%7C7692980eb8e5403c78d108d6a7ed867b%7C8370cf1416f34c16b83c724071654356%7C0&sdata=W%2FcDjt%2F3gmCGZB22SRMJZMI0Zbq7ZzQxnRh38jdJHzU%3D&reserved=0>

[Screenshot from 2019-03-13 14-38-08]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F15019704%2F54309175-a14ef880-459d-11e9-9276-cc3429f4af37.png&data=01%7C01%7Cderek.long%40kcl.ac.uk%7C7692980eb8e5403c78d108d6a7ed867b%7C8370cf1416f34c16b83c724071654356%7C0&sdata=AyGG8EfNm8iRoyLuhBrVMy%2FQCqGNwYgXvTx7bQQCx3A%3D&reserved=0>

keerthanamanivannan, 2019-03-13:
Awesome, what you said made a lot of sense. Thank you for your response!

I changed the domain file to:
[image]
and the `/rosplan_planner_interface` put out a `Plan was solved.` message. 

Thank you once again!

closed 2019-03-13 by keerthanamanivannan