KCL-Planning/ROSPlan #178: forall and when statements in domain file

dr-steveb, 2019-03-12:
I have seen a previous issue asking about the forall keyword, and would like to see if there is any update on a possible implementation for this. Also interested in conditional effects (when). I realize that the forall can be handled with some creative PDDL editing, but not sure about the when. Thanks.

m312z, 2019-03-15:
It's on our todo list (and in the project boards). I hope that it is something that get added in the next cleanup of the codebase which we are starting now.

The "when" effects can also be handled with some creative PDDL editing. For example, but introducing new operators that model the different conditions under which the original conditional operator can be applied.

Michael

corot, 2020-02-16:
To my understanding, all depends on what the user planner supports, right?
By default ROSPlan uses POPF, that does not fully support ADL. But it is possible the change the underlying planner, right?
In my case, I want to explore the usage of disjuntions in the goal (what I don't actually know if exists in PDDL, all said)

m312z, 2020-02-27:
Hello Corot,

Almost all will depend upon the underlying planner, but there are a couple of places it needs support in ROSPlan (and is not implemented yet). I think the major two are:
- Generating the problem file (support for disjunctive goals).
- Applying the (non-sensed) effects of an action in the action interface (support for conditional effects).

Michael

corot, 2020-02-27:
@m312z, thanks for the comment.
What are the currently supported planners? And is POPF the best option among them?
Cannot find anything about using different planners in the documentation.