KCL-Planning/ROSPlan #183: Action dispatched by ROSPlan has a lowercase Action name.

keerthanamanivannan, 2019-03-18:
### Description
I have a simple domain and problem PDDL files, but I'm getting the action dispatch output to be "stow" all in lowercase, but not "Stow" like it's defined in the Domain file. 

### My environment
ROS Distro: Kinetic
OS Version: Ubuntu 16.04
Source build
Branch: master
Git commit: e79d62a
(kinetic branch is far behind master.)

### Steps to reproduce
I cloned ROSPlan and followed few of the tutorials.
I added my own domain and problem PDDL files in rosplan_demos/common/aq/

### Domain file
[image]

### Problem File
[image]

### Actual behaviour
```~/a_ws: rostopic echo /rosplan_plan_dispatcher/action_dispatch 
action_id: 0
name: "stow"
parameters: 
  - 
    key: "a"
    value: "a"
duration: 5.0
dispatch_time: 0.0
---
```
### Expected behaviour
```~/a_ws: rostopic echo /rosplan_plan_dispatcher/action_dispatch 
action_id: 0
name: "Stow"
parameters: 
  - 
    key: "a"
    value: "a"
duration: 5.0
dispatch_time: 0.0
---
```
I am using FlexBE behavior execution engine which takes in the behavior name as is and executes it. 
Let me know how I can fix this.

oscar-lima, 2019-03-18:
The planner itself is (at least one in the pipeline) that converts from uppercase to lowercase (assuming you are using popf). So what you could do is to try and get the source code from popf and modify it, or change the planner for another one.

m312z, 2019-03-18:
Hi both, 

Yes, POPF converts everything to lowercase. Note that PDDL is case insensitive, so this is not exactly incorrect behaviour.

keerthanamanivannan, 2019-03-18:
Alright I see. Thank you for the quick response. 

@m312z Do you know of any other planner which does not convert everything to lowercase, like POPF?
If not, then I will try looking into what @oscar-lima suggested.

oscar-lima, 2019-03-18:
from Michael's answer: PDDL is case insensitive -> you will not find such planner. Although you could hack popf source code not to lowercase your operators, assuming this does not affect the planning process... I think it is cleaner to try and fix on the other side, this means to alter FlexBE code to accept lowercase actions.

keerthanamanivannan, 2019-03-18:
@oscar-lima right!! I totally missed that in Michael's answer. thank you!
I'm trying to make this work by changing the FlexBE behaviors to be in lowercase. Thank you.

closed 2019-03-18 by keerthanamanivannan