KCL-Planning/ROSPlan #187: Problem in changing planner

Pol091994, 2019-04-29:
I'm trying to change planner, from POPF to LPG, so i changed the planner interface in the file interfaced_planning_system.launch: 
[image]

After that i changed the argument of the interface in the planner_interface.launch in: 
[image]

When i launch the services to produce the problem file and the plan i got this error:
[image]

But, if i try to run the same command not via rosplan, but on the linux shell i got no error and the plan is generated.

m312z, 2019-04-29:
It looks like a problem with working directories. I'd guess that the "-out lpgplan" argument is creating the output files in a different directory from "common".

What happens when you run the same command from different directories? Are the files always created in the common folder, in bin/lpg, or in the same directory you run the command?

Pol091994, 2019-04-29:
The files are created in the directory where i run the command

m312z, 2019-04-29:
A quick fix would be to alter the command in the launch file to to place the output files in an absolute path. For example, "-out /home/ai/workspace/src/rosplan/rosplan_planning_system/common/lpgplan"

I can not remember the default working directory of ROS from which a command line will be run. I think it is somewhere in devel. You could find out by locating the lpgplan files.

Best regards,
Michael

oscar-lima, 2019-04-29:
try ~/.ros

Pol091994, 2019-04-29:
Ok perfect, now it works and the planner produce the correct file with the plan in that directory, but the topic planner_output is empty. I think the planner interface doesn't publish the plan. Is this (in planner_interface.launch) enough?
[image]

m312z, 2019-04-30:
That should be enough.
@b-dorian do you know what might be happening here?

b-dorian, 2019-05-01:
Hi, please output the plan to ../rosplan_planning_system/lpgplan.SOL. 

The interface is build to get the plan from that location. 

You can check https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/src/PlannerInterface/LPGPlannerInterface.cpp line 67 to see what is happening behind the hood.

Pol091994, 2019-05-01:
This is the output, the plan is correct (for what i expect). 

[image]

b-dorian, 2019-05-01:
I have tested everything on my end and I get the same result (nothing being published on the topic).

With the popf_planner_interface the plan is correctly published.

I have no idea at the moment why this happens, the code for the publishing is technically identical for both lpg and popf, with only the plan extraction from the text file ( the runPlanner() method) being different. 

I will have a closer look as soon as I am back at the office and get back on this.

Pol091994, 2019-05-02:
ok, perfect. If it is possible have a feedback soon i would be grateful because i need for my degree thesis. Thank you

b-dorian, 2019-05-06:
Refactored the code to work for the refactored version of ROSPlan and removed a bug. Pull ROSPlan from my repo (https://github.com/b-dorian/ROSPlan), or replace the LPG interface code with https://github.com/b-dorian/ROSPlan/blob/master/rosplan_planning_system/src/PlannerInterface/LPGPlannerInterface.cpp.

Also, if you pull the latest version of ROSPlan, please use as -out:  "-out $(find rosplan_planning_system)/test/lpgplan"

Let me know if you encounter any other issues.

m312z, 2019-05-30:
Fixed in #191

closed 2019-05-30 by m312z