KCL-Planning/ROSPlan #188: PDDL questions

keerthanamanivannan, 2019-04-30:
@oscar-lima @m312z I'm turning to you guys' expertise as the last resort after sifting through a bunch of websites that Google spit out for "PDDL example". I would really really appreciate a quick response of any thoughts on how to solve this.

### Background: 
So, most of the things on the internet about PDDL are just basic examples and I have a specific thing which I want to accomplish and I do not know how to move forward for that. 

I have a really basic domain file and a problem file which spits out a single action when given into ROSPlan framework. I get this 
```
rostopic echo /rosplan_plan_dispatcher/action_dispatch
action_id: 0
name: "stow"
parameters: 
  - 
    key: "move_group"
    value: "arms"
duration: 5.0
dispatch_time: 0.0
---
```
when I listen to the action dispatch topic. 

I am giving this output into FlexBE so I created a `rosplan_interface_flexbe` package much like the `rosplan_interface_movebase` but gutted out to conform to FlexBE message type. 

Now. I need to get one more key called **`skip_check`** to be set to **`True`** for any action that ROSPlan is going to spit out. How would I add this parameter (?) to all the actions in the PDDL file? Would I add it as a constant? ANY thoughts on how to achieve this will help. 

Thanks in advance for your time and thoughts! Any links on more detailed PDDL examples would be really helpful too. Thank you!

## Domain File
```(define (domain deep-dive)
(:requirements :typing :durative-actions)

(:types
    arm)

(:predicates
    (stowed ?move_group - arm)
    (unstowed ?move_group - arm)

)
; transformation action
(:durative-action stow
   :parameters (?move_group - arm
   :duration (= ?duration 5)
   :condition (at start (unstowed ?move_group))
   :effect (and (at end (stowed ?move_group))
                (at start (not (unstowed ?move_group))))
)

; transformation action
(:durative-action unstow
  :parameters (?move_group - arm
  :duration (= ?duration 5)
  :condition (at start (stowed ?move_group))
  :effect (and (at end (unstowed ?move_group))
               (at start (not (stowed ?move_group))))
)
)
```
## Problem File
```(define (problem task)
(:domain deep-dive)
(:objects
    arms - arm
)
(:init

    (unstowed arms)

)
(:goal (and
    (stowed arms)
))
)
```

keerthanamanivannan, 2019-04-30:
uh, so I _just_ saw this issue #166. Um. So I cannot add the **`skip_check`** as a constant then?

m312z, 2019-05-01:
Hello,

Does the key have to be a part of the PDDL, or can it be added to the interface?
- If it is only required by FlexBE, and is part of every stow/unstow action, then it can be added in the part of interface that you have written.
- If the interface also requires it as input (for example if you are sending some PDDL action messages from somewhere where `skip_check=False` then it needs to be in the PDDL.

To add the parameter to the PDDL:

First, you are right. PDDL constants are not implemented yet, but it is on our immediate to-do list.

One solution would be to add the parameter as a type instead, by extending the domain to include:
  (:types skip_check)
And the problem file:
  (:objects true - skip_check)
And finally include skip_check as a parameter of stow and unstow. The only possible value this parameter can take is "true".

One issue with this is that it is possible that your interface might reject an action for which skip_check is not set to "true", as no other option exists. In other words, I'm not sure how to define some actions to be valid for dispatch, but not possible for the planner to select (constants would not help with this).

keerthanamanivannan, 2019-05-02:
I see. Yeah. This helps a lot, @m312z. I realized that we need to send this `skip_check` to always be set to True as a part of the stow/unstow action. Which then makes sense for me to add it into the interface. I tried the whole adding it to the PDDL file approach you mentioned, but `rosplan_knowledge_base` node dies for some reason which I can't figure out. 

But you've helped me fix this problem. Thank you for your time, I really appreciate it.

keerthanamanivannan, 2019-05-02:
Closing this issue.

closed 2019-05-02 by keerthanamanivannan