KCL-Planning/ROSPlan #198: bug in knowledge base while querying goals

oscar-lima, 2019-06-11:
From the PR of martinkoling a bug was discovered in the knowledge base, check the following snippet:

```
    ros::NodeHandle nh("~");

    std::string srv_name = "/rosplan_knowledge_base/state/goals";
    ros::ServiceClient client1 = nh.serviceClient<rosplan_knowledge_msgs::GetAttributeService>(srv_name);
    rosplan_knowledge_msgs::GetAttributeService srv;

    srv.request.predicate_name = "anystring";
    ros::service::waitForService(srv_name, ros::Duration(1));
    client1.call(srv);
    // Check that goal attribute_name is "at"
    ASSERT_EQ("at", srv.response.attributes[0].attribute_name);
```

which results in True and should be False

gerardcanal, 2019-06-11:
I believe it's actually the expected behavior. From https://github.com/KCL-Planning/ROSPlan/blob/105cb4cecc6882ec60340219499a60e3ed635078/rosplan_knowledge_base/src/KnowledgeBase.cpp#L539 the request is not considered. I guess the same message as the getPredicate is used for commodity. 

@m312z do you confirm?

oscar-lima, 2019-06-11:
I see... makes sense. Thanks!

closed 2019-06-11 by oscar-lima