KCL-Planning/ROSPlan #202: Repeated VAL files in ROSPlan repo

oscar-lima, 2019-06-16:
e.g. rosplan_dependencies/VAL/include/DebugWriteController.h

vs

rosplan_knowledge_base/src/VALfiles/DebugWriteController.h

m312z, 2019-07-05:
What's the question? We love VAL so we have it twice.

In fact, the version in ROSPlan dependencies is a submodule pointing to the VAL repository (KCL-Planning/VAL) while the version in rosplan_knowledge_base is a much older version that has had the source copied in.

The open task is to remove the VAL  source from rosplan_knowledge_base and use the rosplan_dependencies version instead. There are a few minor differences between the two versions that need to be accounted for.

oscar-lima, 2019-07-05:
Ok, thanks for the clarification!