KCL-Planning/ROSPlan #203: [Q&A] Obtain type of an instance from the KB

dgerod, 2019-06-19:
Hi,

I have added an instance to the KB, for example "it1" of type "item". Later I need to get the type ("item") of an instance("it1") from the KB, would it be possible? After checking the messages and reviewing KB source code, it seems that it is not possible... 

Thanks.

Kind regards,
-- Diego

gerardcanal, 2019-06-19:
Hi Diego!
Hope everything is fine! :) 
You are right, there's no direct way of searching instance's types as for now, but there's a workaround you can do:

You can ask for all the types:
`rosservice call /rosplan_knowledge_base/domain/types "{}"`

And then iterate through the types to get the propositions of each type (empty type name returns all instances):
`rosservice call /rosplan_knowledge_base/state/instances "type_name: 'type'"`

And iterating over the responses to check whether the instance is in that type. 

I know that this is not the best interface to get it, and I believe this is a useful thing to provide. Therefore, I'll flag the issue as an enhancement to be implemented in future versions (soon I hope).

Best regards,
Gerard

dgerod, 2019-06-19:
Hi Gerard,

Yes, all going well thanks. You too, I hope.

Thanks a lot for your help, it works!!! To apply your workaround I have only to write few lines in the [KB interface](https://github.com/dgerod/rosplan_pytools/blob/develop/src/rosplan_pytools/controller/knowledge_base.py)... so, it is a great workaround ;-)

```
def get_instance_type(item_name):

    if not isinstance(item_name, str):
        raise TypeError

    types = _services["get_domain_types"]().types
    for type_name in types:
        instance_names = _services["get_current_instances"](type_name).instances
        if item_name in instance_names:
            return type_name

    return ""
```

Due to our last talk about ROSPlan, I decided to remove the use of MongoDB inside KB interface and create a specific interface to the Scene Database. And this Scene Database is using ROS Server to store the information instead of MongoDB, as you proposed ;-). 

Of course, this is a preliminary version and I have to create a specific node to manage concurrent access to ROS Server parameters related to Scene Database. But if you are interested on that, you could check the code [here](https://github.com/dgerod/rosplan_pytools/blob/develop/src/rosplan_pytools/controller/scene_database.py).

Kind regards,
-- Diego

gerardcanal, 2019-06-19:
Glad it worked!
MongoDB is gone since #197 :-D 
I'll take a look at your scene database, it looks interesting!

dgerod, 2020-06-28:
I have removed [Q&A] prefix as this is not tagged as  this is an enhancement.