KCL-Planning/ROSPlan #217: TFDPlannerInterface takes least optimal plan

varsheim, 2019-09-20:
TFD planner's outputs often more than one file:
tfdplan.1, tfdplan.2, tfdplan.3...
where the last one contains the shortest plan.
TFDPlannerInterface takes only tfdplan.1

varsheim, 2019-09-20:
I will probably fix this in the next few days as i use this Interface

m312z, 2019-11-15:
Hello Varsheim, Thanks for the information!

I know it's been a while since you left the comment - but this is definitely something we should fix. Did you end up making a change? If you a can, it would be great to open a pull request for it.

Michael

varsheim, 2019-11-15:
Hi, i actually did open a pull request some time ago, it is #218 :)
But also i saw that there were some major changes to the popf interface recently and i dont know if the way i set it up is still valid.

oscar-lima, 2019-11-29:
solved by #218

closed 2019-11-29 by oscar-lima