KCL-Planning/ROSPlan #219: Segmentation Fault while planning

marlow-fawn, 2019-10-02:
Hello,

I am new to ROSPlan, and am trying to implement a new domain + problem. When I follow the tutorials but replace the domain and problem files, I get a segmentation fault while planning.

The both pddl files were validated, and worked in other solvers.

Error:
```
[ INFO] [1570040250.544679025]: KCL: (/rosplan_knowledge_base) Parsing domain
[ INFO] [1570040250.544829679]: KCL: (/rosplan_knowledge_base) Parsing domain: /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/cupsworld-s0-compact.pddl.
[ INFO] [1570040250.545814746]: KCL: (/rosplan_knowledge_base) Parsing initial state
[ INFO] [1570040250.545878321]: KCL: (/rosplan_knowledge_base) Parsing Problem File: /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/CW-00-W.pddl.
[ INFO] [1570040250.546385483]: KCL: (/rosplan_knowledge_base) Ready to receive
[ INFO] [1570040250.548662851]: KCL: (/rosplan_problem_interface) Ready to receive
process[message_store-6]: started with pid [20609]
process[replicator_node-7]: started with pid [20610]
process[rosplan_planner_interface-8]: started with pid [20611]
[ INFO] [1570040250.621402990]: KCL: (/rosplan_planner_interface) Ready to receive
[ INFO] [1570040308.504082668]: KCL: (/rosplan_problem_interface) (CW-00-W.pddl) Generating problem file.
[ INFO] [1570040308.857126986]: KCL: (/rosplan_problem_interface) (CW-00-W.pddl) The problem was generated.
[ INFO] [1570040308.857794690]: KCL: (/rosplan_planner_interface) Problem received.
[ INFO] [1570040312.973249119]: KCL: (/rosplan_planner_interface) (CW-00-W.pddl) Writing problem to file.
[ INFO] [1570040312.973622994]: KCL: (/rosplan_planner_interface) (CW-00-W.pddl) Running: timeout 10 /home/marlowfawn/codebase/ROSPlan/src/rosplan/rosplan_planning_system/common/bin/popf /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/cupsworld-s0-compact.pddl /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/CW-00-W.pddl > /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/plan.pddl
timeout: the monitored command dumped core
Segmentation fault
[ INFO] [1570040313.300538958]: KCL: (/rosplan_planner_interface) (CW-00-W.pddl) Planning complete
[ INFO] [1570040313.300745344]: KCL: (/rosplan_planner_interface) (CW-00-W.pddl) Plan was unsolvable.
```

gerardcanal, 2019-10-02:
Hi Marlow,
Can you check what's the output of the following:
`time /home/marlowfawn/codebase/ROSPlan/src/rosplan/rosplan_planning_system/common/bin/popf /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/cupsworld-s0-compact.pddl /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/CW-00-W.pddl > /home/marlowfawn/codebase/ROSPlan/src/macgyver/macgyver_pddl/common/plan.pddl`

At first sight, it looks like POPF is aborting with your input files, so let's discard that.

marlow-fawn, 2019-10-03:
Here's the only output. 
```
Number of literals: 13
Constructing lookup tables: [10%]Segmentation fault (core dumped)
```

gerardcanal, 2019-10-03:
So then it's POPF who is failing and this is not related to ROSPlan. It is possible that your domain is not correct or uses some features not supported by POPF.
For the tutorials, we suggest you use the provided domain and problem files to learn how rosplan works and avoid these kinds of issues.

Then, to try with your domains I would suggest you do the following:

1. Use [VAL](https://github.com/KCL-Planning/VAL) to validate that your domain and problem files are correct.
2. If they are valid (as said by VAL), try another planner that supports your domain file. In the [Planning wiki](https://planning.wiki/) you can find more information on supported files.

Let us know.

marlow-fawn, 2019-10-03:
The issue was with popf not supporting negative preconditions (a part of ADL). Switching over to a ff planner fixed the issue.

closed 2019-10-03 by marlow-fawn