KCL-Planning/ROSPlan #233: rosplan tutorial one as an example make error.

bilal9876, 2020-02-28:
when i try to launch the first tutorial from rosplan tutorials.

[ERROR] [1582850746.648855]: Can't find database at supplied path /home/bilal/rosplan_catkin_ws/src/rosplan/rosplan_knowledge_base/common/mongoDB/. If this is a new DB, create it as an empty directory.
[mongo_server-3] process has died [pid 15958, exit code 1, cmd /home/bilal/rosplan_catkin_ws/src/mongodb_store/mongodb_store/scripts/mongodb_server.py __name:=mongo_server __log:=/home/bilal/.ros/log/8a71736e-59c3-11ea-829a-f496341df899/mongo_server-3.log].
log file: /home/bilal/.ros/log/8a71736e-59c3-11ea-829a-f496341df899/mongo_server-3*.log

i think i should configure the following parametrs in both the strands - mongodb_store and in the rosplan but i really dont know how to integrade the three things together.

so, is there any help or tutorial to go step by step and run melodic with turtlebot3 and rosplan?

oscar-lima, 2020-02-28:
mongoDB is no longer part of ROSPlan implementation, it has been removed in #197. You might want to update your master branch with the latest changes. Regarding the demo with the turtlebot 3 in ROS melodic, it is now located [here](https://github.com/KCL-Planning/rosplan_demos/tree/master/rosplan_turtlebot3_demo). The README however got lost due to some updates but you can still see it [here](https://github.com/KCL-Planning/rosplan_demos/commit/3813d9c66d19a252617c4fd9da1606ceed577f4e?diff=split) (check the right side in green). I will try to fix this over the weekend such that each demo has its own README file inside.

bilal9876, 2020-02-28:
ok.  i downloaded the updated rosplan package and i build it . every thing is going well. thank you very much . i appreciate your help.

closed 2020-02-28 by bilal9876