KCL-Planning/ROSPlan #237: Unable to load type [rosplan_interface_mapping/CreatePRM]

rakhmanu, 2020-04-03:
Hi,

I am following this tutorial: rosplan_turtlebot2_demo and when i run turtlebot_explore.bash script, i have got the following errors which are attached below

[image]

Help me please to solve it
Thank you

oscar-lima, 2020-04-03:
Hi,

CreatePRM is a ROS service provided by the package [rosplan_interface_mapping](https://github.com/KCL-Planning/rosplan_demos/tree/master/rosplan_demos_interfaces/rosplan_interface_mapping)

The error message suggests that you have not built the package before using it or you have not sourced your workspace correctly.

if you are using python catkin tools then you can do: (replace <~/rosplan_ws> with your own path):

```cd <~/rosplan_ws>/src/rosplan_demos/rosplan_demos_interfaces/rosplan_interface_mapping```
```catkin build --this```

then source your workspace (on each terminal):

```source <~/rosplan_ws>/devel/setup.bash```

then to make sure it worked well, do:

```rossrv show CreatePRM```

and you should expect as output:

```
[rosplan_interface_mapping/CreatePRM]:
int32 nr_waypoints
float32 min_distance
float32 casting_distance
float32 connecting_distance
int8 occupancy_threshold
int32 total_attempts
---
string[] waypoints
```

rakhmanu, 2020-04-04:
> Hi,
>
> CreatePRM is a ROS service provided by the package [rosplan_interface_mapping](https://github.com/KCL-Planning/rosplan_demos/tree/master/rosplan_demos_interfaces/rosplan_interface_mapping)
>
> The [...]

thank you It's solved

closed 2020-04-04 by rakhmanu