KCL-Planning/ROSPlan #241: ROSPlan Installation

viraj96, 2020-04-28:
I  am running Ubuntu 18.04 with ROS Melodic. I tried to install ROSPlan as mentioned in the Readme. When I do that I get the following output which ends with errors related to boost,

```
─ catkin build --cmake-args \         
            -DCMAKE_BUILD_TYPE=Release \
            -DPYTHON_EXECUTABLE=/usr/bin/python3 \
            -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m \
            -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic:/usr0/home/vparimi/anaconda3:/usr0/home/vparimi/anaconda3/x86_64-conda_cos6-linux-gnu/sysroot/usr
Workspace:                   /usr0/home/vparimi/Work/Research/software/github/ROSPlan
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Build Space:        [exists] /usr0/home/vparimi/Work/Research/software/github/ROSPlan/build
Devel Space:        [exists] /usr0/home/vparimi/Work/Research/software/github/ROSPlan/devel
Install Space:      [unused] /usr0/home/vparimi/Work/Research/software/github/ROSPlan/install
Log Space:         [missing] /usr0/home/vparimi/Work/Research/software/github/ROSPlan/logs
Source Space:       [exists] /usr0/home/vparimi/Work/Research/software/github/ROSPlan/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.6m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.6m.so
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[build] Found '8' packages in 0.0 seconds.                                                                                                                              
[build] Updating package table.                                                                                                                                         
Starting  >>> catkin_tools_prebuild                                                                                                                                     
Finished  <<< catkin_tools_prebuild                    [ 1.1 seconds ]                                                                                                  
Starting  >>> rosplan_dispatch_msgs                                                                                                                                     
Starting  >>> rosplan_knowledge_msgs                                                                                                                                    
Starting  >>> rosplan_rqt                                                                                                                                               
Starting  >>> rosplan_sensing_interface                                                                                                                                 
Finished  <<< rosplan_sensing_interface                [ 1.5 seconds ]                                                                                                  
Finished  <<< rosplan_rqt                              [ 1.8 seconds ]                                                                                                  
Finished  <<< rosplan_knowledge_msgs                   [ 7.9 seconds ]                                                                                                  
Starting  >>> rosplan_dependencies                                                                                                                                      
Finished  <<< rosplan_dispatch_msgs                    [ 7.9 seconds ]                                                                                                  
________________________________________________________________________________________________________________________________________________________________________
Errors     << rosplan_dependencies:make /usr0/home/vparimi/Work/Research/software/github/ROSPlan/logs/rosplan_dependencies/build.make.000.log                 
In file included from /opt/ros/melodic/include/ros/serialization.h:34:0,
                 from /usr0/home/vparimi/Work/Research/software/github/ROSPlan/devel/.private/rosplan_knowledge_msgs/include/rosplan_knowledge_msgs/KnowledgeItem.h:14,
                 from /usr0/home/vparimi/Work/Research/software/github/ROSPlan/src/rosplan/rosplan_dependencies/ippc_server/include/rosplan_dependencies/ippc_server.h:35,
                 from /usr0/home/vparimi/Work/Research/software/github/ROSPlan/src/rosplan/rosplan_dependencies/ippc_server/src/ippc_server.cpp:1:
/opt/ros/melodic/include/ros/time.h:58:10: fatal error: boost/math/special_functions/round.hpp: No such file or directory
 #include <boost/math/special_functions/round.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ippc_server.dir/ippc_server/src/ippc_server.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ippc_server.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /usr0/home/vparimi/Work/Research/software/github/ROSPlan/build/rosplan_dependencies; catkin build --get-env rosplan_dependencies | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -
........................................................................................................................................................................
Failed     << rosplan_dependencies:make                [ Exited with code 2 ]                                                                                           
Failed    <<< rosplan_dependencies                     [ 10.2 seconds ]                                                                                                 
Abandoned <<< rosplan_knowledge_base                   [ Unrelated job failed ]                                                                                         
Abandoned <<< rosplan_planning_system                  [ Unrelated job failed ]                                                                                         
[build] Summary: 5 of 8 packages succeeded.                                                                                                                             
[build]   Ignored:   1 packages were skipped or are blacklisted.                                                                                                        
[build]   Warnings:  None.                                                                                                                                              
[build]   Abandoned: 2 packages were abandoned.                                                                                                                         
[build]   Failed:    1 packages failed.                                                                                                                                 
[... cut ...]
  File "/usr/local/lib/python3.6/dist-packages/trollius/unix_events.py", line 65, in close
  File "/usr/local/lib/python3.6/dist-packages/trollius/unix_events.py", line 166, in remove_signal_handler
  File "/usr/lib/python3.6/signal.py", line 47, in signal
TypeError: signal handler must be signal.SIG_IGN, signal.SIG_DFL, or a callable object

```

I can see that the boost files are there in `/usr/include/boost/` and cmake does find them but for some reason while building it cant find the header files. Can you help me figure out what might be the issue here?

gerardcanal, 2020-04-28:
Hi @viraj96,
I have not seen this issue before, I am also in 18.04 with Melodic. It seems that it comes from ros headers including boost, but I can't replicate in my machine. The boost files are indeed in `/usr/include/boost/`in my machine too.

Can you confirm you have the file `/usr/include/boost/math/special_functions/round.hpp` which is the one that says is missing? Did you reinstall boost from sources or a different version? Maybe reinstalling boost could fix the issue (I'm using the version from apt).

oscar-lima, 2020-04-28:
There is a similar question asked at ROS answers here: https://answers.ros.org/question/194689/fatal-error-boost/

It might help

viraj96, 2020-04-28:
@gerardcanal I can confirm that `round.hpp` is present in `boost`

```
ls -l /usr/include/boost/math/special_functions/round.hpp
-rw-r--r-- 1 root root 4301 Mar  6  2018 /usr/include/boost/math/special_functions/round.hpp

```

I have `boost` version 1.65.1 and it was installed using `apt`.  I had to define the following environment variables for `catkin` to find my `boost`

```
export BOOST_LIBRARYDIR=/usr/lib/x86_64-linux-gnu                                                                                                               
export BOOST_INCLUDEDIR=/usr/include/boost

```

Without them `cmake` would not have found the libraries like `thread` or `system` etc. 

@oscar-lima I found that question as well when I tried to debug it but I don't specify any `CXX` flags inside my `zshrc` file. Also I am using `zsh` but I don't think that it will cause any issues.

viraj96, 2020-04-28:
@gerardcanal @oscar-lima I was able to solve the issue! The problem was conda. CMake would for some reason just use the conda compiler rather than system ones which caused the whole linking issue in the first place. By forcing CMake to use the system compiler, I was able to resolve the issue and successfully built ROSPlan.

gerardcanal, 2020-04-28:
Glad to read you could solve it! :-)

closed 2020-04-28 by gerardcanal