KCL-Planning/ROSPlan #242: [rosplan_knowledge_base-2] process has been died

rakhmanu, 2020-05-02:
Hi,

I am trying to generate maze PDDL problem by ROSPlan, however when i run roslaunch LaunchFilename.launch, i got the following output, which is attached below

And my domain file is as following:
(define (domain turtlebot_ex)
(:requirements :strips :typing :fluents :disjunctive-preconditions :negative-preconditions)
(:types
	robot  position
)
(:predicates
	(robot_at ?v - robot ?x ?y - position)
	(obstacle ?x ?y) 
    (inc ?a ?b - position)
    (dec ?a ?b - position))
(:action move-up
    :parameters (?v - robot)
    :effect (forall (?x ?y ?yn - position)
                    (when 
                      (and (robot_at ?v ?x ?y)
                           (dec ?y ?yn)
                           (not (obstacle ?x ?yn)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?x ?yn))
                      )
                    )
    )
(:action move-down
    :parameters (?v - robot)
    :effect (forall (?x ?y ?yn - position)
                    (when 
                      (and (robot_at ?v ?x ?y)
                           (inc ?y ?yn)
                           (not (obstacle ?x ?yn)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?x ?yn))
                      )
                    )
     ))
Help me please to fix the error

Thank you in advance
[image]

gerardcanal, 2020-05-05:
Hi @rakhmanu, can you show also the file ´LaunchFilename.launch´?

A reason for this would be that the RDDL knowledge base is loaded instead of the PDDL one.

rakhmanu, 2020-05-06:
Here is the launch file

<launch>
<arg name="gazebo_gui" default="false"/>

<include file="$(find turtlebot_gazebo)/launch/umka_world.launch">
<arg name="gui" value="$(arg gazebo_gui)"/>
</include>

<include file="$(find turtlebot_gazebo)/launch/amcl_demo.launch"/>
<include file="$(find rosplan_interface_mapping)/launch/rosplan_roadmap_server.launch"/>
<include file="$(find rosplan_planning_system)/launch/interfaced_planning_system.launch">
<arg name="domain_path" default="$(find rosplan_demos)/common/navigation-domain.pddl"/>
<arg name="problem_path" value=""/>

<arg name="autom_gen_problem_path" value="$(find rosplan_demos)/common/navigation-problem.pddl"/>

<arg name="data_path" value="$(find rosplan_demos)/common/"/>
</include>
<node pkg="rosplan_interface_movebase" type="rpmovebase" name="rosplan_interface_movebase" respawn="false" output="screen">
<param name="knowledge_base" value="rosplan_knowledge_base"/>
<param name="action_server" value="/move_base"/>
<param name="pddl_action_name" value="goto_waypoint"/>
<param name="action_dispatch_topic" value="/rosplan_plan_dispatcher/action_dispatch"/>
<param name="action_feedback_topic" value="/rosplan_plan_dispatcher/action_feedback"/>
</node>
<node name="rviz" pkg="rviz" type="rviz" args="-d $(find turtlebot_rviz_launchers)/rviz/navigation.rviz"/>
</launch>

And, i used conditional effect (when) in pddl file, so is that possible that the reason of this error is that?

Thank you in advance

rakhmanu, 2020-05-06:
<launch>
<arg name="gazebo_gui" default="false"/>
<include file="$(find turtlebot_gazebo)/launch/umka_world.launch">
<arg name="gui" value="$(arg gazebo_gui)"/>
</include>
<include file="$(find turtlebot_gazebo)/launch/amcl_demo.launch"/>

<include file="$(find rosplan_interface_mapping)/launch/rosplan_roadmap_server.launch"/>

<include file="$(find rosplan_planning_system)/launch/interfaced_planning_system.launch">
<arg name="domain_path" default="$(find rosplan_demos)/common/navigation-domain.pddl"/>

<arg name="problem_path" value=""/>

<arg name="autom_gen_problem_path" value="$(find rosplan_demos)/common/navigation-problem.pddl"/>

<arg name="data_path" value="$(find rosplan_demos)/common/"/>
</include>
<node pkg="rosplan_interface_movebase" type="rpmovebase" name="rosplan_interface_movebase" respawn="false" output="screen">
<param name="knowledge_base" value="rosplan_knowledge_base"/>
<param name="action_server" value="/move_base"/>
<param name="pddl_action_name" value="goto_waypoint"/>
<param name="action_dispatch_topic" value="/rosplan_plan_dispatcher/action_dispatch"/>
<param name="action_feedback_topic" value="/rosplan_plan_dispatcher/action_feedback"/>
</node>
<node name="rviz" pkg="rviz" type="rviz" args="-d $(find turtlebot_rviz_launchers)/rviz/navigation.rviz"/>
</launch>

rakhmanu, 2020-05-06:
I am sorry here is file 
[image]

gerardcanal, 2020-05-06:
Hi rakhmanu,
Thank you for the file, It is possible that this is the case, when the KB terminates like this tends to be due to the parser having problems with the domain. We use VAL's parser for PDDL, so we can only accept what VAL accepts (but I think it should be fine with them, just that rosplan will not integrate -for now- this constructs in the messages, though the planner will receive the full input).

A way to check if your domain is correct is to use VAL. It is installed with rosplan, in ´rosplan_dependencies/VAL´. 
There you have a binary called ´parser´. I checked it with your domain and it states there are errors:

```
$ ./parser domain_rakhmanu.pddl 
File: domain_rakhmanu.pddl

Errors: 3, warnings: 0
domain_rakhmanu.pddl: line: 15: Error: Syntax error in action declaration.
domain_rakhmanu.pddl: line: 15: Error: Unreadable structure
domain_rakhmanu.pddl: line: 16: Error: Syntax error in domain
```
It looks like in your domain the parenthesis in the whens are not correctly placed, so that may be the reason, as the when takes the whole effect, so if the, when does not satisfy the action's effect is empty. Can you try to check the domain and fix it such that the parser states there are no errors, and then check again with rosplan to see if it's resolved?

For info on PDDL you can check here: https://planning.wiki/ref/pddl/

rakhmanu, 2020-05-06:
Dear,

But i could solve correctly PDDL domain and PDDL problem by using online and offline solvers, the only thing is that i cannot implement it in ROSPlan

What can i do ?
Thank you in advance for you help and reply:)

gerardcanal, 2020-05-06:
It is possible that some planners use different parsers and domain representations and they work, or that they can manage to work through some errors in it.
Given that ROSPlan uses VAL parser and domain structure, if VAL parser finds errors in the domain file there are high chances that ROSPlan will malfunction with it. 

The issue may still be there after fixing the domain for VAL, but at least we can discard some possibilities.

rakhmanu, 2020-05-11:
> It is possible that some planners use different parsers and domain representations and they work, or that they can manage to work through some errors in it.
> Given that ROSPlan uses VAL parser and [...]

Okay, i fixed it and I checked PDDL Domain by VAL parser and it says errors 0. However, when i try to integrate it with ROSPlan, still the same error...

gerardcanal, 2020-05-11:
Thanks rakhmanu, can you share the new domain so we can try it?

gerardcanal, 2020-05-11:
Hi again rakhmanu,

I tried with the first domain you sent and I can confirm the parser is finding problems and returning a NULL domain variable, for which then the KB fails (we'll fix that and produce an error in a future release), but the domain file is not correct for the parser.

I can investigate this further in the fixed domain once you share it.

rakhmanu, 2020-05-11:
Hi  @gerardcanal ,
Oh, i am very sorry the domain, that checked without errors, was different 

The parser still shows the errors which indicated the part of forall and when The structure is correct Why it doesn't work? Can you please check it one more time? Does the parser of ROSPlan read conditional effects?

Here is the domain file

(define (domain turtlebot_ex)
(:requirements :strips :fluents :disjunctive-preconditions :negative-preconditions:typing :conditional-effects)
(:types
	robot  position
)
(:predicates
	(robot_at ?v - robot ?x ?y - position)
	(obstacle ?x ?y)
    (inc ?a ?b - position)
    (dec ?a ?b - position))
(:action move-up
    :parameters (?v - robot)
    :effect (forall (?x ?y ?yn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (dec ?y ?yn)
                           (not (obstacle ?x ?yn)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?x ?yn))
                      )
                    )
    )
(:action move-down
    :parameters (?v - robot)
    :effect (forall (?x ?y ?yn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (inc ?y ?yn)
                           (not (obstacle ?x ?yn)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?x ?yn))
                      )
                    )
     )

(:action move-right
    :parameters (?v - robot)
    :effect (forall (?x ?y ?xn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (inc ?x ?xn)
                           (not (obstacle ?xn ?y)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?xn ?y))
                      )
                    )
    )

(:action move-left
    :parameters (?v - robot)
    :effect (forall (?x ?y ?xn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (dec ?x ?xn)
                           (not (obstacle ?xn ?y)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?xn ?y))
                      )
                    )
    )
)

And the parser's output
[image]

Thank you very much

gerardcanal, 2020-05-11:
Hi again,
I see you are missing preconditions in your actions and instead of modelling them with a conditional `when` construct on the effects. I believe it would be better to set those as preconditions to the actions, this way the domain will be easier to read and probably work better. You can check here the syntax for defining actions: https://planning.wiki/ref/pddl/domain#actions 

As long as VAL's parser states there are errors, ROSPlan will not be able to load the domain as the input is not valid PDDL. If the domain is correctly parser, then ROSPlan will integrate it to its supported structures, but input the whole domain to the planner.

rakhmanu, 2020-05-12:
Dear @gerardcanal, 

Thank you very much, now i got errors 0
I just added precondition () below the parameters in every action, and i still have the same errors with knowledge base
[image]

rakhmanu, 2020-05-13:
What can i do for now? Please i need a help

Thank you

gerardcanal, 2020-05-13:
Hi Rakhmanu, I would need the fixed domain to make a last try.

rakhmanu, 2020-05-14:
Hi @gerardcanal ,
Oh thank you very much 
Here is the fixed domain:

(define (domain turtlebot_ex)
(:requirements :strips :fluents :disjunctive-preconditions :negative-preconditions:typing :conditional-effects)
(:types
	robot  position
)
(:predicates
	(robot_at ?v - robot ?x ?y - position)
	(obstacle ?x ?y)
    (inc ?a ?b - position)
    (dec ?a ?b - position))
(:action move-up
    :parameters (?v - robot)
		:precondition()
    :effect (forall (?x ?y ?yn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (dec ?y ?yn)
                           (not (obstacle ?x ?yn)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?x ?yn))
                      )
                    )
    )
(:action move-down
    :parameters (?v - robot)
		:precondition()
    :effect (forall (?x ?y ?yn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (inc ?y ?yn)
                           (not (obstacle ?x ?yn)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?x ?yn))
                      )
                    )
     )

(:action move-right
    :parameters (?v - robot)
		:precondition()
    :effect (forall (?x ?y ?xn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (inc ?x ?xn)
                           (not (obstacle ?xn ?y)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?xn ?y))
                      )
                    )
    )

(:action move-left
    :parameters (?v - robot)
		:precondition()
    :effect (forall (?x ?y ?xn - position)
                    (when
                      (and (robot_at ?v ?x ?y)
                           (dec ?x ?xn)
                           (not (obstacle ?xn ?y)))
                      (and (not (robot_at ?v ?x ?y))
                           (robot_at ?v ?xn ?y))
                      )
                    )
    )
)

gerardcanal, 2020-05-14:
Hello,
With this domain, the KB does not terminate in my machine and VAL says there are not errors.

However, it is still better to add the conditions that you have in the `when` clauses as a precondition, as many planners will not accept that (for instance POPF, which is the default planner used in ROSPlan does not allow conditional effects). It is better to have the conditions for action as a precondition of an operator rather than as conditional effects instead of empty precondition statements. You can find more examples and reference in the IPC benchmark problems, the planning wiki I sent you, in the rosplan_demos packages and in the PDDL book.

I understand the Knowledge Base's process does not die anymore with the fixed domain and therefore it was not an issue in the ROSPlan side, therefore I am closing this one.

closed 2020-05-14 by gerardcanal