KCL-Planning/ROSPlan #245: PDDL parsing failed Plan was unsolvable

rakhmanu, 2020-06-05:
Hi,

I wanted to use my PDDL domain and constructed PDDL problem manually in ROSPlan. However, when i try to execute rosplan_planner_interface, it says plan was unsolvable. 
This is my pddl domain 
(define (domain againd)

(:requirements
        :strips :negative-preconditions 
        :disjunctive-preconditions)
 
(:predicates
        (robot_at ?v   ?wp )
        (adj ?from ?to )
        (obstacle ?wp )
        (visited ?wp )
)
   

(:action goto_waypoint
        :parameters (?v  ?from ?to )
        :precondition (and 
	       (robot_at ?v ?from)
	       (adj ?from?to)
           (not (obstacle ?to))
	       )
        :effect ( and  
	        (visited ?to)
            (not (robot_at ?v ?from))
            (robot_at ?v ?to)
              )
)
   
)

And PDDL problem
(define (problem task)
(:domain againd)
(:objects
    wp0 wp1 wp2 wp3 wp4 wp5
    turtlebot
)
(:init
   
    (robot_at turtlebot wp0)
    (obstacle wp1)
    (adj wp0 wp1)
    (adj wp1 wp0)
    (adj wp1 wp2)
    (adj wp2 wp1)
    (adj wp0 wp3)
    (adj wp3 wp0)
    (adj wp3 wp4)
    (adj wp4 wp3)
    (adj wp4 wp5)
    (adj wp5 wp4)
    (adj wp1 wp4)   
    (adj wp4 wp1)  
    (adj wp2 wp5)
    (adj wp5 wp2)
     
)
(:goal (and
    
    (visited wp5)
))
)

When i check PDDL model by VAL parser, there is no error with domain but there are warnings in problem file:
Errors: 0, warnings: 4
againp2.pddl: line: 9: Warning: Undeclared symbol: robot_at
againp2.pddl: line: 10: Warning: Undeclared symbol: obstacle
againp2.pddl: line: 11: Warning: Undeclared symbol: adj
againp2.pddl: line: 29: Warning: Undeclared symbol: visited

By the way, my launch file 

 
[image]

How i can fix this?

oscar-lima, 2020-06-05:
This domain does not have durative actions and therefore standard popf can't solve it. I suggest you try ffha. sudo apt install ros-melodic-ffha

You can see an example on this open PR:

https://github.com/KCL-Planning/rosplan_demos/pull/33/files

closing for now, feel free to reopen if further issues

closed 2020-06-05 by oscar-lima

rakhmanu, 2020-06-08:
Hi,

Thank you very much for the reply
I tried to use FF planner with ffha by following this https://github.com/KCL-Planning/rosplan_demos/pull/33/files example. However, it still says that the plan was unsolvable...

The pddl domain and problem from the examples in /ROSPlan/rosplan_planning_system/test/pddl/amazon

 Here is the FFplanning launch file : 

<?xml version="1.0"?>
<launch>

	
	<arg name="node_name"            default="rosplan_planner_interface" />
	<arg name="use_problem_topic"    default="true" />
	<arg name="problem_topic"        default="/rosplan_problem_interface/problem_instance" />
	<arg name="planner_topic"        default="planner_output" />
	<arg name="domain_path"          default="$(find ROSPlan/rosplan_planning_system/test/pddl/amazon/domain2.pddl" />
	<arg name="problem_path"         default="$(find ROSPlan/rosplan_planning_system/test/pddl/amazon/problem2.pddl" />
	<arg name="data_path"            default="$(find rosplan_planning_system)/test/pddl/amazonrosplan_planner_interfa/" />
	<arg name="planner_command"      default="timeout 10 rosrun ffha ffha -o DOMAIN -f PROBLEM" />
	<arg name="planner_interface"    default="ff_planner_interface" />
	<arg name = "use_ffha"           default="true" />
	
	<node name="$(arg node_name)" pkg="rosplan_planning_system" type="$(arg planner_interface)" respawn="false" output="screen">

		
		<param name="use_problem_topic" value="$(arg use_problem_topic)" />

		
		<param name="problem_topic" value="$(arg problem_topic)" />
		<param name="planner_topic" value="$(arg planner_topic)" />

		
		<param name="domain_path" value="$(arg domain_path)" />
		<param name="problem_path" value="$(arg problem_path)" />
		<param name="data_path" value="$(arg data_path)" />

		
		<param name="planner_command" value="$(arg planner_command)" />
		<param name="use_ffha" value="$(arg use_ffha)" />
		
	</node>

</launch>

And my launch file:
<?xml version="1.0"?>
<launch>
     
    <include file="$(find rosplan_planning_system)/launch/includes/rosplan_planner_interfaces.launch"  >

    <arg name="domain_path"	default="$(find rROSPlan/rosplan_planning_system/test/pddl/amazon/domain2.pddl" />
    <arg name="problem_path" default="$(find ROSPlan/rosplan_planning_system/test/pddl/amazon/problem2.pddl" />

    </include>
        
  

</launch>

[image]