KCL-Planning/ROSPlan #246: Automatic PDDL problem generation

rakhmanu, 2020-06-05:
Hi,

I have a PDDL domain and i want to generate automatically Problem file from it, however ROSPlan generates an empty problem from the knowledge base. How to generate PDDL problem instances automatically from KB? 
This is the domain file
(define (domain againd)

(:requirements
        :strips :negative-preconditions 
        :disjunctive-preconditions)
 
(:predicates
        (robot_at ?v   ?wp )
        (adj ?from ?to )
        (obstacle ?wp )
        (visited ?wp )
)
   

(:action goto_waypoint
        :parameters (?v  ?from ?to )
        :precondition (and 
	       (robot_at ?v ?from)
	       (adj ?from?to)
           (not (obstacle ?to))
	       )
        :effect ( and  
	        (visited ?to)
            (not (robot_at ?v ?from))
            (robot_at ?v ?to)
              )
)
   
)

Thank you very much in advance

m312z, 2020-06-05:
Hi there,

What has been loaded into the knowledge base? It is not clear what you are expecting.

What are the objects and initial sate that you are expecting to produce? The Knowledge Base can store the PDDL state and produce problem files in PDDL syntax, but it is not creating new information. The sensing interface can update the knowledge base with new information (from sensors).

Michael

oscar-lima, 2020-06-05:
Have you tried the rosplan rqt interface? 

rqt --standalone rosplan_rqt.dispatcher.ROSPlanDispatcher

That offers a gui interface where you can populate your KB with objects.

Another alternative is to load an initial pddl problem to knowledge base, this is controlled by:
problem_path variable, located under: https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/launch/interfaced_planning_system.launch#L10

by default is empty, but you can set a custom file

Closing for now, feel free to reopen if issue persists

closed 2020-06-05 by oscar-lima