KCL-Planning/ROSPlan #252: robot does not move

rakhmanu, 2020-06-29:
Hi

I am using FF planner and gave my own waypoints to the robot to follow waypoints.yaml file
I have changed the rosplan_roadmap_server.launch file to rpsimplemapServer. And the problem was generated correctly and the plan was solved. There is an error that says argument list too long, it happened after adding instances to the initial state+goals knowledge base in explore.bash file
  
I attached the output from terminal 

Thank you for you help) 

Sincerely, Ulzhalgas Rakhman

[image]
[image]

rakhmanu, 2020-06-30:
I solved the error shown above but robot still does not move...

oscar-lima, 2020-06-30:
Can you please comment on how you solved this issue? (for others to know). Also the robot not moving seems like an issue that would fit better the rosplan demos repository.

rakhmanu, 2020-07-07:
The solution was to delete the added argument attribute name in explore_common.bash

rakhmanu, 2020-07-07:
@oscar-lima  Hi
I have got another error which says that waypoint not found in parameter server as i am using rpsimplemapserver , the data is derived from mongodb not parameter server. It means weneed to rewrite rpsimplemapserver, is that right?
Thank you

closed 2020-11-10 by rakhmanu