KCL-Planning/ROSPlan #254: FFPlannerInterface is not able to work with Contingent-FF

dgerod, 2020-06-30:
Although it is written that ff_planner_interface works with Contingent-FF, it is not working well. The problem is that it returns that the plan is unsolvable when it is solvable.

After Contingent-FF generates correctly generating the plan, the "FFPlannerInstance" should look for "ff: found plan as follows" 
to check if the plan was solved or not. However, it is looking for "ff: found legal plan". See below:

```
bool solved = false;
while (not solved and std::getline(planfile, line)) {
	// skip lines until there is a plan
	if (line.find("ff: found legal plan") != line.npos) {
		solved = true;
	}
}
```

Looking for "ff: found legal plan" is correct because it is what FF planner writes, but it is not what Contingent-FF does. So, "FFPlannerInstance" should also look for "ff: found plan as follows".

dgerod, 2020-07-05:
#255 issue should be consider together with this one.

dgerod, 2020-07-11:
Working on this issue, see: https://github.com/dgerod/ROSPlan/tree/feature/Add-ContingentFF-planner

dgerod, 2020-08-01:
Pull request open #256.

orhaimwerthaim, 2021-04-11:
I like your system and the great tutorials. Can you please add a working example for contingent planning?

dgerod, 2021-04-18:
Sure.

Domain file:
```
(define (domain boxes)
(:requirements :typing :disjunctive-preconditions)

(:types
    place
    object
    bin - place
    box - place
	gear - object
)

(:predicates
	(robot-at ?p - place)
	(free-tool)

	(object-at ?o - object ?p - place)
	(localised ?o - object)
	(grasped ?o - object)
)

(:action move_to_place
	:parameters
	    (?p1 - place
	     ?p2 - place
	    )
	:precondition
	    (and (robot-at ?p1)
	         (not(robot-at ?p2))
        )
	:effect
		(and (robot-at ?p2)
	         (not(robot-at ?p1))
	    )
)

(:action find_object
	:parameters
	    (?c - bin
	     ?o - object
	    )
	:precondition
	    (and (free-tool)
	    	 (robot-at ?c)
	         (not (localised ?o))
	    )
	:effect
	    (and (robot-at ?c)
	         (localised ?o)
			 (object-at ?o ?c)
	    )
)

(:action pickup_object
	:parameters
	    (?c - bin
	     ?o - object
	    )
	:precondition
	    (and (robot-at ?c)
	         (free-tool)
	         (localised ?o)
	         (object-at ?o ?c)
	         (not(grasped ?o))
	    )
	:effect
	    (and (robot-at ?c)
	         (not(free-tool))
	         (grasped ?o)
	         (not (object-at ?o ?c))
	    )
)

(:action drop_object
	:parameters
	    (?b - box
	     ?o - object
	    )
	:precondition
	    (and (robot-at ?b)
	         (not(free-tool))
	         (grasped ?o)
	    )
	:effect
	    (and (robot-at ?b)
	         (free-tool)
	         (not(grasped ?o))
			 (object-at ?o ?b)
	    )
)

)
```

Problem file:
```
(define (problem task)
(:domain boxes)
(:objects
    home - place
    bin_2 - bin
    box_1 - box
    gear_1 gear_2 gear_3 gear_4 - gear
)
(:init
    (robot-at home)

    (object-at gear_1 bin_2)
    (object-at gear_2 bin_2)
    (object-at gear_3 bin_2)
    (object-at gear_4 bin_2)

    (free-tool)
)
(:goal (and
    (object-at gear_1 box_1)
    (object-at gear_2 box_1)
    (object-at gear_3 box_1)
    (object-at gear_4 box_1)
))
)
```

orhaimwerthaim, 2021-04-18:
Thank you very much for your reply. 
Can you please add the **launch file** (I need a **full example**)? 
I was not able to connect the contingent-FF planner to ROSPlan.
More specifically, it could not parse the plan generated by the planner.

In my example, I used the PDDL extension (words 'observe', 'unknown', and 'oneof') for contingent planning with partially observable domains. 

See https://github.com/KCL-Planning/rosplan_demos/blob/master/rosplan_demos/common/d.pddl
and https://github.com/KCL-Planning/rosplan_demos/blob/master/rosplan_demos/common/p.pddl

Thanks,
Or

dgerod, 2021-04-18:
Are you using FF PlannerInterface with Contingent-FF? The output file of Contingent-FF planner is different than the one of FF planner, and this is the reason why it could not be used with FF PlannerInterface. You need to use the code that I wrote in PR #256 as I said above, the PR is not merge to ROSPlan.

orhaimwerthaim, 2021-04-19:
Thank you!!!.

One last thing, When I am planning, and action failed (or something like that), I want to perform replanning.
I am currently doing it using a module I wrote that triggers ROSPlan's services (planning, parsing, and dispatching). Does ROSPlan have this kind of control module integrated into it (as can be understood from the ROSPlan article)?

dgerod, 2021-04-20:
No, there is not other way to do that, what you are doing is the way to do re-planning. 

In a previous version of ROSPlan it was possible to configure an automatic re-planning when an action failed, but this functionality was removed. Maybe the article you are reading is the first one of ROSPlan where the system is explained, this article is based on that version that had automatic re-planning.

orhaimwerthaim, 2021-04-21:
Thank you for your fast and helpful replies.

LiamMZ, 2023-12-28:
@dgerod Would you be able to provide an example with uncertainty in the initial state/ limited observability of the initial locations of the objects that need to be manipulated? I am also having trouble being able to use Contingent-FF to the fullest.

Thank you!

dgerod, 2024-01-03:
Hi @LiamMZ, I don't understand you request, I am not an expert in planning. My work is not in planners but on how execute the generated actions, so my PDDL files are quite simple. Sorry for not be able to help you.