KCL-Planning/ROSPlan #255: PDDLSimplePlanParser is not able to parse Contingent-FF

dgerod, 2020-07-04:
The parsed plan from Contingent-FF is not being generated by PDDLSimplePlanParser. 

And it is normal because the parser is expected a syntax as POPF planner:
`0.000: (goto_place b1)  [0.001]`

While Contingent-FF output to be parser is:
`0||0 --- GOTO_PLACE C1 --- SON: 1||-1`

And I think that this is not working also with FF, its syntax is:
`[0: PICKUP_OBJECT C1 V1]`

In [documentation](https://kcl-planning.github.io/ROSPlan//documentation/interfaces/02_planner_interface.html) is written that planner interface works with FF, Metric-FF and Contingent-FF among others. So, I have understood that all nodes of ROSPlan work with Contingent-FF, but they are not working. And is FF being used by someone?

Therefore, in my opinion this is a bug or missed functionality. Could you confirm that my understanding is correct? Thanks.

Finally, I would like to talk about the design of "planning_interface" and "parse_plan nodes". Parsing a planner output is hard coupled to the planner itself, so why these two functionalities are split in two different nodes ("planner_interface" and "parse_plan")? In my opinion they should be done by only one node that is the planner_interface, and it should generate a ROSPlan defined format of actions. In addition, I think that having the "planner_command" parameter for "planner_interface" is not a good idea because it is also hard coupled with implementation of the planner (e.g. POPF, FF). The only information that is not couple so they should be pass through parameters is the timeout and the directory path where the planner binary is stored.

dgerod, 2020-07-05:
After checking with more detail I understand that the FFPlannerInterface transforms the output of the FF planner to POPF plan format. So, the PDDLSimplePlanParser is able to parse the results of FF planner. This transformation to POPF plan format is what I named as ROSPlan defined format.

Therefore, the problem of parsing the Contingent-FF is not related to PDDLSimplePlanParser but to the FFPlannerInterface, and it is related to #254. This should be solved by transforming the Contingent-FF plan file to POPF plan format in FFPlannerInterface.

dgerod, 2020-07-11:
Working on this: https://github.com/dgerod/ROSPlan/tree/feature/Add-ContingentFF-planner

dgerod, 2020-08-01:
Pull request #256.

bilal9876, 2020-08-01:
> The parsed plan from Contingent-FF is not being generated by PDDLSimplePlanParser. And it is normal because the parser is expected a syntax as POPF planner:
> 
> `0.000: (goto_place b1) [0.001]`
> 
> While Contingent-FF output to be parser is:
> 
> `0||0 --- GOTO_PLACE C1 --- SON: 1||-1`
> 
> And I think that this is not working also with FF, its syntax is:
> 
> `0: PICKUP_OBJECT C1 V1`
> 
> In the [documentation](https://kcl-planning.github.io/ROSPlan//documentation/interfaces/02_planner_interface.html) is written that planner interface works with FF, Metric-FF and Contingent-FF among others. So, my understanding is that all nodes of ROSPlan work with Contingent-FF what is not happening.
> 
> Therefore, in my opinion this is a bug. Could you confirm that my understanding is correct? Thanks.
> 
> Finally, I would like to talk about the design of "planning_interface" and "parse_plan nodes". Parsing a planner output is hard coupled to the planner itself, so why these two functionalities are split in two different nodes? In my opinion they should be done by only one node that is the "planner_interface", and it should generate a ROSPlan defined format of actions.
> 
> In addition, I think that having the "planner_command" parameter for "planner_interface" is not a good idea because it is also hard coupled with implementation of the planner (e.g. POPFPlannerInterface). The only information that is not couple to the planner is the timeout and the directory path where the planner binary is stored, so these should be passed to the node as parameters.

bilal9876, 2020-08-01:
in particular, a contingent plan is a tree structure and after the ROSPlan converts it to a ROS message action by parsing interface, is there any objectives to create a contingent plan.

dgerod, 2020-08-03:
@bilal9876, sorry but I cannot get your point.

bilal9876, 2020-08-03:
I mean that a contingency plan is a plan where the effect of any action may result in 'x' state (AND branch) or in 'y' state (OR branch) when the contingency plan is converted to temporal plan, (what is implemented in ROSPlan framework), the temporal plan considers only one branch (AND branch) as the effect of the action. So, after parsing the plan and then dispatching the plan, when action fail or the result of the action is not the same as the  predicted one, the ROSPlan framework will replan again from the scratch and not will consider the another branch (OR branch) of the previously contingent plan.

is this true or not ?

m312z, 2021-04-01:
Hello Bilal,

You are correct, the esterel/simple plan parsing assumes non-contingent plans. For parsing contingent plans you should use the PNP parser, which will generate a petr-net-plan for dispatch - you can find that here:
https://sites.google.com/site/rosplantopnp/home

About separating the planner interface and plan parsing nodes - this is how it should be. If they were one node, this couples the plan representation to the planner, which is not always desired. For example, I might want to generate plans with POPF and represent them as ESTEREL, PNP, or some other format. In addition, this allows to  publish the plan as string to be shown to the user, or connected to some other component (not plan parsing and dispatch).

Michael

closed 2021-04-01 by m312z

dgerod, 2021-04-03:
@m312z , one question about this issue. I would like to understand the procedure.

This issue is related to a missed functionality not with the question you are answering. I opened a PR to fixing it some time ago but it is still not merged, so is it correct to close this issue?