KCL-Planning/ROSPlan #259: convert the plan into a PDF:

bilal9876, 2020-07-19:
I generated the plan.dot from the terminal by executing: " rostopic echo /rosplan_plan_dispatcher/plan_graph"

I copied the data from the terminal and i removed the "data:" keyword from the plan

when i tried to convert the plan into pdf by using:   dot -Tpdf plan.dot > plan.pdf

"Error: plan.dot: syntax error in line 103 near "

 i don't know what is the problem, really i try to modify the generated plan several times, but no solution.

rakhmanu, 2020-11-02:
Hi, 
Please remove all "\n"  in plan graph 
Then it will work 
Thank you

m312z, 2021-04-01:
Hello both,

This is to do with how the topic is subscribed to. When using rostopic echo, use the argument "-p" for pretty-print. This will format correctly to your screen/file and you'll not have an issue with DOT.

Michael

m312z, 2021-04-01:
Gerard has a script for generating PDF, which uses this line:
`rostopic echo /rosplan_plan_dispatcher/plan_graph -n 1 -p | sed "N;s/.*digraph/digraph/g" | dot -Tpdf > $OUTPUT` 
Here's the fulle script (which we can add to the repo)
```
#!/bin/bash
# Usage: ./show_plan.sh <path_to_plan>.pdf
# i.e. ./show_plan.sh /tmp/my_plan.pdf

OUTPUT="/tmp/plan.pdf"
if [ ! -z "$1" ]; then
    OUTPUT=$1
fi
rostopic echo /rosplan_plan_dispatcher/plan_graph -n 1 -p | sed "N;s/.*digraph/digraph/g" | dot -Tpdf > $OUTPUT
xdg-open $OUTPUT &
echo "Esterel plan has been printed in $OUTPUT"
```

closed 2021-04-01 by m312z

gerardcanal, 2021-04-01:
Just added the script in a PR (#283)