KCL-Planning/ROSPlan #262: Problem Generation fails due to unavailable services of the knowledge base

MeFisto94, 2020-10-07:
So I'm trying to generate/preprocess a problem, which makes use of metrics, functions and fluents. 
Apparently this leads to the following log output:
```
1602068948.947029358 INFO /rosplan_problem_interface [/build/turtlebot/src/ROSPlan/rosplan_planning_system/src/ProblemGeneration/ProblemInterface.cpp:128(ProblemInterface::runProblemServer)] [topics: /rosout, /rosplan_problem_interface/problem_instance] KCL: (/rosplan_problem_interface) (out_problem.pddl) Generating problem file.
1602068949.171313681 ERROR /rosplan_problem_interface [/build/turtlebot/src/ROSPlan/rosplan_planning_system/src/ProblemGeneration/PDDLProblemGenerator.cpp:150(PDDLProblemGenerator::makeInitialState)] [topics: /rosout, /rosplan_problem_interface/problem_instance] KCL: (PDDLProblemGenerator) Failed to call service /rosplan_knowledge_base/domain/functions
1602068949.171767171 ERROR /rosplan_problem_interface [/build/turtlebot/src/ROSPlan/rosplan_planning_system/src/ProblemGeneration/PDDLProblemGenerator.cpp:209(PDDLProblemGenerator::makeGoals)] [topics: /rosout, /rosplan_problem_interface/problem_instance] KCL: (PDDLProblemGenerator) Failed to call service /rosplan_knowledge_base/state/goals
1602068949.172151532 ERROR /rosplan_problem_interface [/build/turtlebot/src/ROSPlan/rosplan_planning_system/src/ProblemGeneration/PDDLProblemGenerator.cpp:268(PDDLProblemGenerator::makeMetric)] [topics: /rosout, /rosplan_problem_interface/problem_instance] KCL: (PDDLProblemGenerator) Failed to call service /rosplan_knowledge_base/state/metric
1602068949.172228339 INFO /rosplan_problem_interface [/build/turtlebot/src/ROSPlan/rosplan_planning_system/src/ProblemGeneration/ProblemInterface.cpp:130(ProblemInterface::runProblemServer)] [topics: /rosout, /rosplan_problem_interface/problem_instance] KCL: (/rosplan_problem_interface) (out_problem.pddl) The problem was generated.
```

From the tutorials, I was not able to infer if I need to launch the knowledge base in any special way to support that, but seeing that at least calls to makeGoals succeed for others with "non fluent" files, makes me wonder if something is bad on my side.
Does anyone have some insight in how I could debug this further?

On a sidenote: Such an error does not prevent the generator from producing a plan, it just lacks a lot of content.

Thanks in Advance

gerardcanal, 2020-10-07:
Hi @MeFisto94,
The problem generator will probably go on even when service calls fail and generate a problem with the information it has.

To check a bit why this may happen, did you change the name of the knowledge base node? In such case, you need to specify it in the problem generator launch too (in the "knowledge_base" argument).

If you didn't change it, can you check which services are the knowledge base providing? I.e. `rosservice list | grep knowledge` and check if those services are there. If they are not there, make sure the knowledge base was able to run. If your domain cannot be parsed, the knowledge base will not be able to start and may die.

MeFisto94, 2020-10-07:
Apparently the services _are there:
```
# rosservice list | grep knowledge
/rosplan_knowledge_base/clear
/rosplan_knowledge_base/domain/functions
/rosplan_knowledge_base/domain/name
/rosplan_knowledge_base/domain/operator_details
/rosplan_knowledge_base/domain/operators
/rosplan_knowledge_base/domain/predicate_details
/rosplan_knowledge_base/domain/predicates
/rosplan_knowledge_base/domain/types
/rosplan_knowledge_base/get_loggers
/rosplan_knowledge_base/query_state
/rosplan_knowledge_base/set_logger_level
/rosplan_knowledge_base/state/functions
/rosplan_knowledge_base/state/goals
/rosplan_knowledge_base/state/instances
/rosplan_knowledge_base/state/metric
/rosplan_knowledge_base/state/propositions
/rosplan_knowledge_base/state/timed_knowledge
/rosplan_knowledge_base/update
/rosplan_knowledge_base/update_array
/rosplan_knowledge_base/update_constraints_oneof
/rosplan_knowledge_base/update_sensed_predicates
```

So the name is `rosplan_knowledge_base', and that's reflected in the launch file:
```xml
# cat tutorial_02.launch 
<?xml version="1.0"?>
<launch>

    <!-- arguments -->
    <arg name="domain_path"     default="/python/pddl/domain.pddl" />
    <arg name="problem_path"    default="/python/pddl/problem.pddl" />

    <!-- knowledge base -->
    <node name="rosplan_knowledge_base" pkg="rosplan_knowledge_base" type="knowledgeBase" respawn="false" output="screen">
        <param name="domain_path" value="$(arg domain_path)" />
        <param name="problem_path" value="$(arg problem_path)" />
        <!-- conditional planning flags -->
        <param name="use_unknowns" value="false" />
    </node>

    <!-- problem generation -->
    <include file="$(find rosplan_planning_system)/launch/includes/problem_interface.launch">
        <arg name="knowledge_base"   value="rosplan_knowledge_base" />
        <arg name="domain_path"      value="$(arg domain_path)" />
        <arg name="problem_path"     value="/python/pddl/out_problem.pddl" />
        <arg name="problem_topic"    value="problem_instance" />
    </include>
    <!-- planner interface -->
    <include file="$(find rosplan_planning_system)/launch/includes/planner_interface.launch">
        <arg name="use_problem_topic"    value="true" />
        <arg name="problem_topic"        value="/rosplan_problem_interface/problem_instance" />
        <arg name="planner_topic"        value="planner_output" />
        <arg name="domain_path"          value="$(arg domain_path)" />
        <arg name="problem_path"         value="/python/pddl/out_problem.pddl" />
        <arg name="data_path"            value="/python/pddl" />
        <arg name="planner_command"      value="timeout 10 $(find rosplan_planning_system)/common/bin/popf DOMAIN PROBLEM" />
    </include>
</launch>
```

Now the full logs show something, though:
The knowledge base died (even though the services are listed) and the rosplan_planner_interface calls my domain file faulty (type errors).

```pddl
; Numeric planning in practice is rather limited, because planners are limited, even though the spec is not.
; One of these issues is the cost optimization. For Fast-Downward, the cost metric needs to be called "total-cost"
; This domain used "total-distance" before, but apparently that is as illegal as using ":fluents" in the requirements
(define (domain mail)
    (:requirements :typing :negative-preconditions :equality :action-costs) ; Other synonyms for action costs  :numeric-fluents :fluents :action-costs)

    (:types location - object
            letter - object
    )

    (:constants mailbox - location)

    (:predicates
        (at ?l - location); If the Salesman is at the given location
        (canDropAt ?let - letter ?loc - location) ; Can the given letter be dropped at the location loc.
        ; this even allows us to drop letters at various locations
        (deliver ?let - letter) ; Used for the goal "deliver all letters".
        (batched ?let - letter) ; TODO: Use to simulate the stack.
    )

    (:functions
        (distance ?from ?to) - number
        ; Fast-Downward accepted (batch-count) without anything
        (batch-count) - number
        (total-cost) - number
    )

    (:action goto
        :parameters (?from - location ?to - location)
        :precondition (and
            (at ?from)
            (not(at ?to)) ; Prevent useless loops/from == to
        )
        :effect (and
            (at ?to)
            (not (at ?from))
            (increase (total-cost) (distance ?from ?to))
        )
    )

    ; This is explicitly modelled, so that we can model things like recharging pauses
    ; TODO: Currently there is no modelling of the stack content, and as such, this method won't be used yet.
    ; The Planner has no reason to pick up letters.
    (:action pickup
        :parameters (?let - letter)
        :precondition (and
            (at mailbox)
            ; TODO: The following line is not supported by fast downward
            (<= (batch-count) 5)
        )
        :effect (and
            ; TODO: The following line is not supported by fast downward
            (increase (batch-count) 1)
        )
    )

    (:action drop
        :parameters (?let - letter ?loc - location)
        :precondition (and
            (at ?loc)
            (canDropAt ?let ?loc)
            ; TODO: The following line is not supported by fast downward
            (> (batch-count) 0)
        )
        :effect (and
            (deliver ?let)
            ; TODO: The following line is not supported by fast downward
            (decrease (batch-count) 1)
        )
    )
)
```

Is there anything obviously wrong? 
A lot of planners were rejecting it because the fast downward translator does not support action costs, whereas the web editor (planning.domains?) did work with domain and plan but disregarding statements like the batch-count.
Unfortunately I cannot check the knowledgeBase's log file, as the stated file does not exist.

gerardcanal, 2020-10-07:
The services probably are there because the knowledge base registered them with master, but if you try to call them it will fail.

I think that the increase and decrease constructs may not be supported but I'm not 100% sure. ROSPlan PDDL Knowledge Base uses VAL's parser, therefore we can only parse whatever VAL can parse (and we are using version 1.2 of VAL at the moment).

I would suggest trying with VAL's parser, which is bundled in the rosplan_dependencies package (rosplan_dependencies/VAL). There's an executable there called parser and add the path to your domain file as an argument to see if there are any errors. 

If the domain cannot be parsed, it sometimes generates an error and the knowledge base dies. I believe this is probably what's happening here.

MeFisto94, 2020-10-07:
`rosrun rosplan_dependencies parser domain.pddl problem.pddl` (and without the problem.pddl) both give me `Errors: 0, warnings: 0`.

Is there any way to find out why the knowledge base dies? I mean apparently it looks like it dies, I just don't know why and don't have any logs.

It seems I forgot to show the relevant log actually:
```
 INFO] [1602084582.843699615]: KCL: (/rosplan_problem_interface) (out_problem.pddl) Generating problem file.
[ERROR] [1602084583.062649760]: KCL: (PDDLProblemGenerator) Failed to call service /rosplan_knowledge_base/domain/functions
[ERROR] [1602084583.063106562]: KCL: (PDDLProblemGenerator) Failed to call service /rosplan_knowledge_base/state/goals
[ERROR] [1602084583.063390482]: KCL: (PDDLProblemGenerator) Failed to call service /rosplan_knowledge_base/state/metric
[ INFO] [1602084583.063456186]: KCL: (/rosplan_problem_interface) (out_problem.pddl) The problem was generated.
[ INFO] [1602084583.063590560]: KCL: (/rosplan_planner_interface) Problem received.
[rosplan_knowledge_base-1] process has died [pid 756, exit code -11, cmd /build/turtlebot/devel/lib/rosplan_knowledge_base/knowledgeBase __name:=rosplan_knowledge_base __log:=/root/.ros/log/52cbda74-08ac-11eb-be66-0242ac110002/rosplan_knowledge_base-1.log].
log file: /root/.ros/log/52cbda74-08ac-11eb-be66-0242ac110002/rosplan_knowledge_base-1*.log
``

gerardcanal, 2020-10-07:
Ok, I see now. That's a bit stranger as when it's a parsing problem the KB dies at the start when it parses the domain and problem files. This seems like something different. Can you add an example problem file and the commands you run? I will try to reproduce it on my end.

But before that, could you try to launch everything and call the services manually from `rosservice call`? Straight away before planning or generating any problem. That would help to see if that's actually some issue with the services and know which one.

MeFisto94, 2020-10-07:
```pddl
(define (problem shortpath)
    (:domain mail)
    (:requirements :typing :negative-preconditions :equality :action-costs :numeric-fluents) ; Other synonyms for action costs  :numeric-fluents :fluents :action-costs)
    (:objects F E G - location
        A B C D - letter
    )
    (:init
        (canDropAt A F)
        (canDropAt B E)
        (canDropAt C G)
        (canDropAt D G)
        (canDropAt D F)

        (at mailbox)
        (= (total-cost) 0)
        (= (distance F E) 1)
        (= (distance F G) 2)
        (= (distance E G) 4)
        ; Bidirectional routes available
        (= (distance E F) 1)
        (= (distance G F) 2)
        (= (distance G E) 4)

        ; Required by Fast Downward and maybe others, even though it's impossible/illegal:
        (= (distance F F) 0)
        (= (distance E E) 0)
        (= (distance G G) 0)
        (= (distance mailbox mailbox) 0)

        (= (distance F mailbox) 1)
        (= (distance mailbox F) 1)
        (= (distance E mailbox) 2)
        (= (distance mailbox E) 2)
        (= (distance G mailbox) 2)
        (= (distance mailbox G) 2)

        (= (batch-count) 4) ; All four letters. TODO: Remove as soon as pickup is modelled.
    )

    ; Fast Downward - Goal needs to stand BEFORE metric
    (:goal (and (deliver A) (deliver B) (deliver C) (deliver D)))
    (:metric minimize (total-cost))
)
```

The command I run is a script:
```bash
screen -m -A -d -S tut roslaunch tutorial_02.launch
sleep 5
rosservice call /rosplan_problem_interface/problem_generation_server
rosservice call /rosplan_planner_interface/planning_server
```

The tutorial_02.launch is pasted a bit further up but should be equal to the actual tutorial.
I should also mention this part of the log, but this happens only when executing the planner and as such might be related to a broken knowledge base and not the cause:
```
[ INFO] [1602092782.950766252]: KCL: (/rosplan_planner_interface) (out_problem.pddl) Running: timeout 10 /build/turtlebot/src/ROSPlan/rosplan_planning_system/common/bin/popf /python/pddl/domain.pddl /python/pddl/out_problem.pddl > /python/pddl/plan.pddl
-----------------------------------------------------

The supplied domain/problem file appear to violate part of the PDDL
language specification.  Specifically:

The planner will continue, but you may wish to fix your files accordingly
Type Errors Encountered in Domain File
--------------------------------------

Due to type errors in the supplied domain file, the planner
has to terminate.  The log of type checking is as follows:

```
The only type error I could imagine is the `- number` which is no declared type.

Manual calls before generating the problem work partially:
```
# rosservice call /rosplan_knowledge_base/state/metric
metric: 
  knowledge_type: 3
  initial_time: 
    secs: 0
    nsecs:         0
  is_negative: False
  instance_type: ''
  instance_name: ''
  attribute_name: ''
  values: []
  function_value: 0.0
  optimization: "minimize"
  expr: 
    tokens: 
      - 
        expr_type: 1
        constant: 0.0
        function: 
          name: "total-cost"
          typed_parameters: []
        op: 0
        special_type: 0
  ineq: 
    comparison_type: 0
    LHS: 
      tokens: []
    RHS: 
      tokens: []
    grounded: False
```

```
 rosservice call /rosplan_knowledge_base/domain/functions 
ERROR: transport error completing service call: unable to receive data from sender, check sender's logs for details
```
After that the service is down and calling metric again just yields "Unable to communicate with service".

Trying it again after restarting:
` rosservice call /rosplan_knowledge_base/state/goals "predicate_name: ''" ` also works.
Multiple calls to both /state's work.

Calling `rosservice call /rosplan_knowledge_base/domain/functions` is what actually causes the death of the knowledge base.

gerardcanal, 2020-10-08:
Hello @MeFisto94,
I found the issue! The error was in the domain file, and when getting the information from the functions (this uses VAL) there was a segmentation fault in VAL's code. The error was hen processing the functions and trying to get the types of the parameters, in a parameter called ?from.

Thus, the issue was in the functions declaration, that was not specifying the type of the parameters in the distance function. It should be like this:

```
(distance ?from - location ?to - location) - number
```
With this, the problem generation works well:
```
[ INFO] [1602144999.242057601]: KCL: (/rosplan_problem_interface) (out_problem.pddl) Generating problem file.
[ INFO] [1602144999.429396541]: KCL: (/rosplan_problem_interface) (out_problem.pddl) The problem was generated.
```

The planning server also works, though a solution can't be found. You will have to check with popf that a solution can be found (as rosplan just uses it as a planner), or use a planner that can deal with your domain. For instance, popf does not support negative preconditions, which appear in your domain (see https://planning.wiki/ref/planners/popf) for admitted requirements.

I understand the issue is solved so I'm closing it, please feel free to reopen if that's not the case.

closed 2020-10-08 by gerardcanal

MeFisto94, 2020-10-08:
Hey, I'll give that a try and see what I can do. Do you know any planner that would support negative preconditions and numeric fluents other than popf that is integrated with rosplan? This time it's probably trivial to live without negative preconditions, but I don't know if that applies to all my future problems.

Is this something that, in theory, would be worth reporting upstream to VAL? I'm also curious why `parse` did not segfault in that case.

Anyway thanks for your always swift replies and the help looking into it. I'm going to check and re-open the issue in case I cannot get it to work with that :)

gerardcanal, 2020-10-08:
Hello,
I believe Metric-FF may work in your case, and it's also supported by ROSPlan.

VAL's latest version is different to the one in ROSPlan, and what it's not supporting it is the popf planner. Each planner supports what they support, but sometimes there are workarounds. For instance, for the negative preconditions you may have two predicates per each predicate: itself and itself negated. I.e. you may have `(at ?robot waypoint)` and `(not-at ?robot ?waypoint)`. Both are positive predicates, but act as the negative one. It may be a bit trickier to maintain as you need to update them correctly.