KCL-Planning/ROSPlan #263: Rosplan Plan "Plan was unsolvable" error

HadeelJazzaa, 2020-10-24:
Both of 
[image]
[image]
[image]
[image]
[image]

pddl files were validated and worked in other solvers but when It comes with the popf planner I get a segmentation fault while planning!.
 Can you help me in this Issue please?

gerardcanal, 2020-10-26:
Hi @HadeelJazzaa,
I don't think POPF can handle this domain, as I don't believe it works with numerical comparisons as a precondition. You can use ROSPlan with many solvers (and even add your own). Check [the ROSPlan tutorials](http://kcl-planning.github.io/ROSPlan//tutorials/tutorial_02) for help in doing this, so you can use a solver that works for your domain.

I'm closing the issue as it's not directly ROSPlan-related but the domains not working with POPF (we use popf but do not maintain it, as well as the other planners).

closed 2020-10-26 by gerardcanal

HadeelJazzaa, 2020-10-26:
Thanks Gerard
Yes I think  that as well.
I am not specialist in planning. So, what I am looking for is a suggestion of what planner can handle this domain?

If you recommend any!

gerardcanal, 2020-10-26:
Hello,
I have not used those kinds of domains, so I am not 100% sure. It seems like OPTIC (which is the successor of popf) may be able to. You mentioned you used some solvers already for that domain, so if you have any that works already. I believe you need some planner that supports the `:numeric-fluents` requirement. You can check the [planning.wiki](https://planning.wiki) for more information on planners and what they support: https://planning.wiki/ref/pddl21/req#numeric-fluents. 

Hope this helps!

HadeelJazzaa, 2020-10-26:
Ok
OPTIC sounds good.
Thank you so much Gerard for the help.

Regards.

DerekLong101, 2020-10-26:
Actually, there is not a problem with POPF using comparison preconditions. The requirements are ill-specified here, though - you don't need disjunctive preconditions or durative actions for this domain. If you send me the domain source and problem source I'll work out why it isn't solving - it is something simple, I think.

Cheers

Derek

HadeelJazzaa, 2020-10-26:
Thanks a lot Derek,

I've attached the files.

Pleas have a look.

Best Regards.

Hadeel

University of Huddersfield inspiring global professionals.
[http://marketing.hud.ac.uk/_HOSTED/EmailSig2014/EmailSigFooterMarch2019.jpg]

This transmission is confidential and may be legally privileged. If you receive it in error, please notify us immediately by e-mail and remove it from your system. If the content of this e-mail does not relate to the business of the University of Huddersfield, then we do not endorse it and will accept no liability.

DerekLong101, 2020-10-26:
Maybe mailing through github strips them off - mail me directly at derek.long@kcl.ac.uk.

Cheers

Derek

[http://marketing.hud.ac.uk/_HOSTED/EmailSig2014/EmailSigFooterMarch2019.jpg]