KCL-Planning/ROSPlan #264: KCL: (RPActionInterface) could not call Knowledge Base for operator details, move_forward

rakhmanu, 2020-11-02:
Hi, 

I wrote PDDL for the navigation and by using FF planner it was solved. After parsing, the planner output produces the following errors: 
[ERROR] [1604296110.545727969, 172.890000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, turn_left
[ERROR] [1604296110.546277400, 172.890000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, move_forward
[ERROR] [1604296110.546890748, 172.891000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, move_forward
[ERROR] [1604296110.547383708, 172.891000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, move_forward
[ERROR] [1604296110.547920246, 172.892000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, turn_right
[ERROR] [1604296110.548421968, 172.892000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, move_forward
[ERROR] [1604296110.548946863, 172.893000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, move_forward
[ERROR] [1604296110.549455599, 172.893000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, turn_left
[ERROR] [1604296110.550065310, 172.894000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, move_forward
[ERROR] [1604296110.550583426, 172.895000000]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for operator details, turn_right

Why the actions cannot be found in Knowledge Base for the parsing process?

Thank you
Sincerely, Ulzhalgas Rakhman

gerardcanal, 2020-11-02:
Hi Ulzhalgas,
This seems to be the parsing interface not being able to call the operator details service from the knowledge base. A reason could be that the knowledge base has died and can not respond. 

Can you make sure the knowledge base is alive (i.e. with `rosnode info`) and if it is try to call the `operator_details` service of the knowledge base with the different operators when this error happens and see what does the terminal shows?

rakhmanu, 2020-11-02:
@gerardcanal , thank you for the reply

[image]

In all operators the same error ... 
And when i type this, "rosservice call / rosplan_knowledge_base/domain/ operators", it says unable to communicate with service...

gerardcanal, 2020-11-02:
Hello,
Sorry I was not clear. To check if the Knowledge Base is alive, you can run:

`rosnode info /rosplan_knowledge_base`

To check if the operator is there, you can run:
`rosservice call /rosplan_knowledge_base/domain/operator_details "name: 'move_forward'"`

And if this answers something like `ERROR: service [/rosplan_knowledge_base/domain/operator_details] responded with an error: ` please call the following:
`rosservice call /rosplan_knowledge_base/domain/operators "{}"`

And make sure the move_forward operator appears in the list of operators. If it does not appear, it means that the domain has not loaded the operator.

rakhmanu, 2020-11-04:
Hi @gerardcanal ,

Thank you for the quick reply
As you suggested i checked if the Knowledge Base alive and it gives me the error: Communication with node failed. 
All operator checking by  rosservice call /rosplan_knowledge_base/domain/operator gives error. 
And i could not find a knowledge base, when i run rqt
But added initial state and goals to knowledge base and loaded domain, problem files to KB
However, still cannot call a rosplan_knowledge_base
Is there something else that i did not follow?

gerardcanal, 2020-11-04:
Hello,
It seems then that the knowledge base is having an error and failing. Can
you try to find the error message saying something like
`[rosplan_knowledge_base-2] process has died` and check what's the output
before that?

It may be that there's something in the domain that is not correctly parsed
by the knowledge base, and sometimes this generates a segfault.

Missatge de rakhmanu <notifications@github.com> del dia dc., 4 de nov. 2020
a les 6:12:

rakhmanu, 2020-11-05:
Hi @gerardcanal ,
Yes, i can find the error which says [rosplan_knowledge_base-16] process has died [pid 28950, exit code -11]. The output before that is:
[ INFO ] [1604539017.846207474]: Physics dynamic reconfigure ready.
Before this output i have one more error ( [ERROR] [1604539017.822064290, 8296.493000000]: KCL: (RPActionInterface) could not call Knowledge Base for operator details, move_forward
 ) which appeared before calling the services. 
As you said, i will parse each predicate one by one to check is that the domain parsed correctly.

Thank you

rakhmanu, 2020-11-05:
I solved the issue. RPSimulatedActionInterface was used, so it was changed to RPActionInterface

closed 2020-11-05 by rakhmanu