KCL-Planning/ROSPlan #266: PDDL without "types"  causes errors

rakhmanu, 2020-11-09:
Hi,

I noticed that the parsing is impossible if i use PDDL domain and problem (FF planner),  where "types -(:typing in requirements)" is unused.  I mean i always need to used "types" in problem files
Is that possible to fix it?

Thank you
Sincerely, Ulzhalgas Rakhman

gerardcanal, 2020-11-09:
Hi Ulzhalgas,
Can you be a bit more specific on what does it mean that it is impossible? What errors do they appear?

rakhmanu, 2020-11-09:
I meant that this PDDL is okay for parsing
(define (domain blocks_domain)

(:requirements :strips :typing :fluents :disjunctive-preconditions )

(:types block_t)

(:predicates 
	(inhand ?block - block_t) 
	(emptyhand) 
	(not_emptyhand) 
	(on ?block ?on_block - block_t) 
	(clear ?block - block_t)
)

(:action pick_up
    :parameters (?block ?from_block - block_t)
    :precondition (and 
		 (emptyhand)
		 (clear ?block)
		 (on ?block ?from_block))
    :effect (and 
		 (inhand ?block)
		 (clear ?from_block)
		 (not (emptyhand))
		 (not_emptyhand)
	     (not (on ?block ?from_block)))
)

(:action put_down
    :parameters (?block ?on_block - block_t)
    :precondition (and 
		 (not_emptyhand)
		 (clear ?on_block)
		 (inhand ?block))
    :effect (and 
		 (on ?block ?on_block)
		 (emptyhand)
		 (not (not_emptyhand))
		 (not (inhand ?block))
		 (not (clear ?on_block)))
  )
)

But this (is shown below) cannot be parsed in ROSPlan, i checked it by using VAL
Does it depend on the planner?

(define (domain blocks_domain)

(:requirements :strips :fluents :disjunctive-preconditions )

(:predicates 
	(inhand ?block ) 
	(emptyhand) 
	(not_emptyhand) 
	(on ?block ?on_block ) 
	(clear ?block )
)

(:action pick_up
    :parameters (?block ?from_block )
    :precondition (and 
		 (emptyhand)
		 (clear ?block)
		 (on ?block ?from_block))
    :effect (and 
		 (inhand ?block)
		 (clear ?from_block)
		 (not (emptyhand))
		 (not_emptyhand)
	     (not (on ?block ?from_block)))
)

(:action put_down
    :parameters (?block ?on_block)
    :precondition (and 
		 (not_emptyhand)
		 (clear ?on_block)
		 (inhand ?block))
    :effect (and 
		 (on ?block ?on_block)
		 (emptyhand)
		 (not (not_emptyhand))
		 (not (inhand ?block))
		 (not (clear ?on_block)))
  )
)

gerardcanal, 2020-11-09:
Thank you. I am still not sure about what is the error that rosplan is
showing. Is the knowledge base dying?

Gerard Canal

Missatge de rakhmanu <notifications@github.com> del dia dl., 9 de nov. 2020
a les 9:59:

rakhmanu, 2020-11-09:
Yes, if i dont use types in pddl, it says knowledge base died

gerardcanal, 2020-11-10:
Hi @rakhmanu,
I had some time to check your domains and you were right, there was an issue where domains without types would produce a segfault in the Knowledge Base. This has been fixed in #269 and will be merged soon. 

Thank you for spotting this!

PR #269 (2020-11-10): Fix #266
Fixed #266 where untyped domains would segfault in case the domain was not typed. Set the default type to "object"
commit ca0cdc50 (2020-11-10): Fix #266
commit ff3d1eb1 (2020-11-10): Merge pull request #269 from gerardcanal/fix_266

Fix #266
closed 2020-11-10 by m312z