KCL-Planning/ROSPlan #270: Replanning

AngeloFerrando, 2020-11-30:
I am having problems understanding how the replan mechanism works. I looked into ROSPlan papers, documentation and tutorials, but I have not find an example to help me understand it.
Specifically, it is not clear to me if the replanning is handled inside ROSPlan, or if it is a feature I can obtain with low effort (but that I have to implement). According to some other issues raised, I see that I could obtain replanning by catching the failure of a plan, and asking (using services) to plan again.
But, I also read about 'Filter Violations'. If I understood correctly, ROSPlan can automatically synthesise conditions to check at runtime (starting from pre/post-conditions in the domain/problem file). Is it right? Does it mean that, as long as I update its knowledge base, ROSPlan will replan when a violation of pre/post-conditions happen? If yes, do you have any example/tutorial that I could have a look at?
Thank you in advance.

gerardcanal, 2020-11-30:
Hi Angelo,
ROSPlan does not provide any automatic replanning mechanism. However, it should be easy to implement. When an action fails, it may be up to the user to decide how to recover that failure, and that may or may not be up to the planning. ROSPlan provides the components that can be put together to get planning and execution, but we intend to provide flexibility and not a standard use.

Thus, once an action fails, action dispatch stops and the easiest replanning method is to just go over the planning loop again (problem generation, planning, dispatching), repeating this while the goals have not been achieved (the dispatcher responds with that in DispatchService.srv)

Regarding the checks, you are correct. If you use the abstract Action Interface, it will check that the preconditions are correct before executing the action, and (I believe that it also checks) that the postconditions have been satisfied. In any of the two cases, the dispatch will stop. Generating a new problem from the KB (assuming it's correctly updated). 

In summary, ROSPlan will not replan by itself, but you can trigger the replan in the same way you trigger the initial planning, and ROSPlan will use the information in the KB when you call the problem generation service. So, provided that the KB is updated, it will create a problem according to the current state of the world. If you are using a robot, you can check the [sensing interface](http://kcl-planning.github.io/ROSPlan//tutorials/tutorial_11) for info on automatic updating of the KB based on sensors.

I hope this answer helps, feel free to ask any further doubts. @m312z do you want to add anything?

m312z, 2020-11-30:
Hello both,

Gerard is right - ROSPlan provides the components for the system, but does not dictate how they are put together to form some system behaviour.

For example, you could launch various nodes (knowledge base, problem generation, planner and parsing nodes, plan dispatch, sensing and action interfaces) and then use a bash/python script with the following psuedocode:
```
# while not goals_achieved:
  # call generate problem and plan services
  # if problem is unsolvable return failure
  # goals_achieved = call dispatch service
  # if goals_achieved, return success
```
I think it would be very useful to provide a tutorial on this - the turtlebot3 demo should contain the necessary scripts already.

AngeloFerrando, 2020-11-30:
Thank you both for your clarification.