KCL-Planning/ROSPlan #276: Planner is not able to solve the problem if we have negative conditions in problem file

harshkakashaniya, 2021-01-14:
Planner is not able to plan if I have a negative condition in Problem file.
for example:
(not (docked kenney))

Does not work. 
Am I doing something wrong? Any solutions?

harshkakashaniya, 2021-01-15:
Error in terminal .
-----------------------------------------------------

The supplied domain/problem file appear to violate part of the PDDL
language specification.  Specifically:

The planner will continue, but you may wish to fix your files accordingly
A problem has been encountered, and the planner has to terminate.
-----------------------------------------------------------------
Unfortunately, at present, the planner does not fully support ADL
unless in the rules for derived predicates.  Only two aspects of
ADL can be used in action definitions:
- forall conditions, containing a simple conjunct of propositional and
  numeric facts;
- Conditional (when... ) effects, and then only with numeric conditions
  and numeric consequences on values which do not appear in the
  preconditions of actions.

To use this planner with your problem, you will have to reformulate it to
avoid ADL.  Alternatively, if you have a particularly compelling case
for them, please contact the authors to discuss it with them, who may be able to
extend the planner to meet your needs.

gerardcanal, 2021-01-15:
Duplicate of #176 

Hello @harshkakashaniya,
POPF does not support negative conditions. You should use another planner or change your domain file. 

ROSPlan interfaces with an external planner, and therefore we cannot fix issues with the domain or support of the domains by those external planners. Hope this helps.

closed 2021-01-15 by gerardcanal

harshkakashaniya, 2021-01-15:
Thanks gerardcanal...
I also went through the #176 answer.
You guys are right it makes sense to had 2 variables in some cases as it is duratives. 
For example:
attach and unattach instead of just attach and not(attach) 

when the system is in the transition phase of from attach to unattach. Both the flags will be false.

Thank you once again.