KCL-Planning/ROSPlan #280: Action dispatch only when previous action is complete for non-durative actions

mdmdmd9876, 2021-03-15:
Hi, 

I wanted to ask for some clarification regarding the esteral-plan dispatcher. 
I noticed that for non-durative operators, all actions are being dispatched at the same time, even if the previous action has not finished. Another thing is that there are no edges between action1_END --> action2_START of two consecutive action nodes action1 and action2 but only between the two start nodes action1_Start --> action2_START. 

Is there any way to enforce dispatching of actions only when the previous one has finished, even when using non-durative pddl operators?

Thanks in advance!

jack2200, 2021-04-01:
i have the same problem the key words action_duration have no function

m312z, 2021-04-07:
@mdmdmd9876 it sounds as though you want to be using the "simple dispatcher" instead. This other dispatcher dispatches action sequentially - waiting for the previous one to complete before dispatching the next.

With the esterel disaptcher, instant actions are represented purely by the start node of a durative action. The end node still appears in the esterel plan and is drawn in the graph - but it is not used for anything. All of the conditions and effects of the instant action are assumed to be in the start node.

We'll open up a PR soon to prune away those end nodes, and represent the action with a single "instant action" node instead.

Note that this probably won't solve your problem - since the action is instant, then the next action is not going to wait for it to "complete" before executing. In case any of its preconditions are sensed predicates, then the next action will wait for those precondtions to be achieved. Sensed predicates can be set by updating the knowledge base - the sensing interface does this automatically for any predicates is updates through sensing. However, in any other situation the next action will not wait.

In case you really do need the next action to wait for the first one to finish - I suggest that the action is not really an instant action in the first place.

Michael