KCL-Planning/ROSPlan #282: Installing ROSPlan on Ubuntu 20.04 with ROS neotic

erfanshojaei, 2021-03-25:
Hello,

I am trying to install RosPlan on Ubuntu 20.04. ROS neotic has been already installed on Ubuntu 20.04. 
After running the "_catkin build_"  command , I have got the following error:

******************************************************************************************************************
---------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/erfan/ROSPlan
--------------------------------------------------------
Build Space:        [exists] /home/erfan/ROSPlan/build
Devel Space:        [exists] /home/erfan/ROSPlan/devel
Install Space:      [unused] /home/erfan/ROSPlan/install
Log Space:          [exists] /home/erfan/ROSPlan/logs
Source Space:       [exists] /home/erfan/ROSPlan/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
[build] Found '8' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> rosplan_dispatch_msgs                                            
Starting  >>> rosplan_knowledge_msgs                                           
Starting  >>> rosplan_rqt                                                      
Starting  >>> rosplan_sensing_interface                                        
Finished  <<< rosplan_sensing_interface                [ 0.2 seconds ]         
Finished  <<< rosplan_rqt                              [ 0.2 seconds ]         
Finished  <<< rosplan_dispatch_msgs                    [ 2.3 seconds ]         
Finished  <<< rosplan_knowledge_msgs                   [ 2.0 seconds ]         
Starting  >>> rosplan_dependencies                                             
_______________________________________________________________________________
Warnings   << rosplan_dependencies:cmake /home/erfan/ROSPlan/logs/rosplan_dependencies/build.cmake.003.log
CMake Warning (dev) at CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/erfan/ROSPlan/build/rosplan_dependencies; catkin build --get-env rosplan_dependencies | catkin env -si  /usr/bin/cmake /home/erfan/ROSPlan/src/rosplan/rosplan_dependencies --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/erfan/ROSPlan/devel/.private/rosplan_dependencies -DCMAKE_INSTALL_PREFIX=/home/erfan/ROSPlan/install; cd -
...............................................................................
_______________________________________________________________________________
Errors     << rosplan_dependencies:make /home/erfan/ROSPlan/logs/rosplan_dependencies/build.make.000.log
/home/erfan/ROSPlan/src/rosplan/rosplan_dependencies/rddl_parser/task_analyzer.cc: In member function ‘void TaskAnalyzer::analyzeTask(int, int)’:
/home/erfan/ROSPlan/src/rosplan/rosplan_dependencies/rddl_parser/task_analyzer.cc:29:43: error: implicitly-declared ‘State& State::operator=(const State&)’ is deprecated [-Werror=deprecated-copy]
   29 |             currentState = State(nextState);
      |                                           ^
In file included from /home/erfan/ROSPlan/src/rosplan/rosplan_dependencies/rddl_parser/task_analyzer.h:4,
                 from /home/erfan/ROSPlan/src/rosplan/rosplan_dependencies/rddl_parser/task_analyzer.cc:1:
/home/erfan/ROSPlan/src/rosplan/rosplan_dependencies/rddl_parser/states.h:22:5: note: because ‘State’ has user-provided ‘State::State(const State&)’
   22 |     State(State const& other) : state(other.state) {}
[... 85 lines cut ...]
      |     ^~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/rddl_parser.dir/build.make:140: CMakeFiles/rddl_parser.dir/rddl_parser/task_analyzer.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:207: CMakeFiles/rddl_parser.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/erfan/ROSPlan/build/rosplan_dependencies; catkin build --get-env rosplan_dependencies | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -
...............................................................................
Failed     << rosplan_dependencies:make                [ Exited with code 2 ]  
Failed    <<< rosplan_dependencies                     [ 10.7 seconds ]        
Abandoned <<< rosplan                                  [ Unrelated job failed ]
Abandoned <<< rosplan_knowledge_base                   [ Unrelated job failed ]
Abandoned <<< rosplan_planning_system                  [ Unrelated job failed ]
[build] Summary: 4 of 8 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 3 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 12.9 seconds total._   

******************************************************************************************************************
Could you please kindly help me to resovle this issue?

Thank you in advance and regards,
Erfan

gerardcanal, 2021-03-25:
Hi Erfan,
We are in the process of migrating to ROS Noetic. Please try this PR #268 in the meantime. This one should work.

Best,
Gerard

closed 2021-03-25 by gerardcanal

Karan-BEHAR-REBLADE, 2024-08-15:
Hello @gerardcanal 

While installing the ROSPlan for ROS2 I receive the following errors. 

vagrant@ubuntu:~/rosplan_ws$ sudo apt install flex bison freeglut3-dev libbdd-dev python-catkin-tools ros-$ROS_DISTRO-tf2-bullet
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python-catkin-tools

Please advise of the further steps. Bien coridalement