KCL-Planning/ROSPlan #290: Cannot run prost

orhaimwerthaim, 2021-04-21:
Hellow, 
I followed the [ROSPlan's RDDL toturial](https://kcl-planning.github.io/ROSPlan//tutorials/tutorial_12) and wanted to test the prost planner as written in the tutorial.

I was running:
$ ./run_prost.sh created_rddl_domain.rddl instance.rddl "[PROST -s 1 -se [IPPC2014]]"

And got the following error:
Error: Could not find or load main class rddl.competition.Server

Can you please guide me on how to solve this issue?
Additionally, if you have, I would be happy to get a link to a working example of ROSPlan with RDD. Mainly I would like to see a full launch file.

Thanks a lot!

gerardcanal, 2021-04-21:
Hello @orhaimwerthaim,
I can't reproduce the issue, as the run_prost.sh works well for me. The class it says it can not load is from rddlsim. Rddlsim is shipped with rosplan under rosplan_dependencies and it should be compiled with catkin_make. Please make sure this has been compiled during the catkin build. Also, I believe you need to run the run_prost.sh file from the folder it is in for it to work well.

Regarding examples, here there are some: https://github.com/gerardcanal/SoPS but the launchfiles are probably modified and it's not the standard case as this was used to perform some experiments.

orhaimwerthaim, 2021-04-21:
Thank you!
I will rebuild prost on my machine.

gerardcanal, 2021-04-21:
Note it's not prost, it's rddlsim which is inside rosplan_dependencies. The post binary should be fine, and should work well with rosplan's parser. I need to update it to the new version but I have not had time yet. But your error is on in prost.

gerardcanal, 2021-04-23:
Please let us know if that fixed the issue. I am closing it now, but feel free to reopen it if still unsolved.