KCL-Planning/ROSPlan #293: Metric-FF fails to find solution

rakhmanu, 2021-05-26:
Hi,

When I try to use Metric-FF in RosPlan, it always fails to find a solution.  When I check the PDDL files by VAL in RospLan, it gives no error. 
However,  I installed the Metric-FF planner from an external source, and it works well. 

This s the launch file:

<include file="$(find rosplan_planning_system)/launch/includes/planner_interface_metricff.launch"> 
		<arg name="use_problem_topic"    value="false" />
		<arg name="planner_topic"        value="planner_output" />
		<arg name="domain_path"          value="$(find try)/common/PDDL/domain.pddl" />
		<arg name="problem_path"         value="$(find try)/common/PDDL/problem.pddl" />
		<arg name="data_path"            value="$(find try)/common/PDDL/" />
		<arg name="planner_command"      value="timeout 10 $(find rosplan_planning_system)/common/bin/Metric-FF DOMAIN PROBLEM" />
	</include>

This is planner_interface_metricff.launch

<?xml version="1.0"?>
<launch>

	
	<arg name="node_name"            default="rosplan_planner_interface" />
	<arg name="use_problem_topic"    default="true" />
	<arg name="problem_topic"        default="/rosplan_problem_interface/problem_instance" />
	<arg name="planner_topic"        default="planner_output" />
	<arg name="domain_path"          default="$(find try)/common/PDDL/generated_d.pddl"  />
	<arg name="problem_path"         default="$(find try)/common/PDDL/generated_p.pddl"  />
	<arg name="data_path"            default="$(find try)/common/" />
	<arg name="planner_command"      default="timeout 10 $(find rosplan_planning_system)/common/bin/Metric-FF DOMAIN PROBLEM" />
	<arg name="planner_interface"    default="ff_planner_interface" />
	
	<node name="$(arg node_name)" pkg="rosplan_planning_system" type="$(arg planner_interface)" respawn="false" output="screen">

		
		<param name="use_problem_topic" value="$(arg use_problem_topic)" />

		
		<param name="problem_topic" value="$(arg problem_topic)" />
		<param name="planner_topic" value="$(arg planner_topic)" />

		
		<param name="domain_path" value="$(arg domain_path)" />
		<param name="problem_path" value="$(arg problem_path)" />
		<param name="data_path" value="$(arg data_path)" />

		
		<param name="planner_command" value="$(arg planner_command)" />
		
	</node>

</launch>

Am I missing something or something is wrong? 
Could you please have a look?

Thank you very much, 
Sincerely Ulzhalgas Rakhman

domain.pddl

(define (domain tetris)
(:requirements :typing :equality :negative-preconditions :action-costs)
  (:types  
	one_square two_straight right_l - pieces
	position 	
	)

  (:predicates
    (clear ?xy - position)
[... 335 lines cut ...]
(and
(clear f0-0f)
(clear f0-1f)
(clear f0-2f)
(clear f0-3f)
(clear f1-0f)
(clear f1-1f)
(clear f1-2f)
(clear f1-3f)
(clear f2-0f)
(clear f2-1f)
(clear f2-2f)
(clear f2-3f)
(clear f3-0f)
(clear f3-1f)
(clear f3-2f)
(clear f3-3f)
(clear f4-0f)
(clear f4-1f)
(clear f4-2f)
(clear f4-3f)
)
)
(:metric minimize (total-cost))
)

rakhmanu, 2021-05-26:
Hi @gerardcanal,
Can you please use the Metric-FF included in rosplan_planning_system/common/bin to make sure it works with your files?
---> Yes I am using Metric-FF included in rosplan_planning_system/common/bin

The output of the plan is this:

usage of ff:

OPTIONS   DESCRIPTIONS

-p <str>    Path for operator and fact file
-o <str>    Operator file name
-f <str>    Fact file name

-r <int>    Random seed [used for random restarts; preset: 0]

-s <int>    Search configuration [preset: s=5]; '+H': helpful actions pruning
      0     Standard-FF: EHC+H then BFS (cost minimization: NO)
      1     BFS (cost minimization: NO)
      2     BFS+H (cost minimization: NO)
      3     Weighted A* (cost minimization: YES)
      4     A*epsilon (cost minimization: YES)
      5     EHC+H then A*epsilon (cost minimization: YES)
-w <num>    Set weight w for search configs 3,4,5 [preset: w=5]

-C          Do NOT use cost-minimizing relaxed plans for options 3,4,5

-b <float>  Fixed upper bound on solution cost (prune based on g+hmax); active only with cost minimization

gerardcanal, 2021-05-26:
I think I see the issue now.
The planner command is not correct. You need to specify the flags correctly. I believe it should be like this (sorry I can't test it at the moment):
`timeout 10 $(find rosplan_planning_system)/common/bin/Metric-FF -o DOMAIN -f PROBLEM`

The planner command line must be the same you use as in the command line, but instead of writing the domain and problem pat you need to change those for the strings `DOMAIN` and `PROBLEM` so that ROSPlan can then replace those flags for the correct paths to the domain and generated problem files.

rakhmanu, 2021-05-26:
As you suggested @gerardcanal I changed 
'''
<include file="$(find rosplan_planning_system)/launch/includes/planner_interface_metricff.launch"> 
   	<arg name="use_problem_topic"    value="false" />
   	<arg name="planner_topic"        value="planner_output" />
   	<arg name="domain_path"          value="$(find try)/common/PDDL/domain.pddl" />
   	<arg name="problem_path"         value="$(find try)/common/PDDL/problem.pddl" />
   	<arg name="data_path"            value="$(find try)/common/PDDL/" />
   	<arg name="planner_command"      value="timeout 10 $(find rosplan_planning_system)/common/bin/Metric-FF -o DOMAIN -f PROBLEM" />
   </include>
'''
It still says that "Plan was unsolvable". But, plan.pddl file is empty now...

m312z, 2021-05-26:
Hello both,

I notice that "timeout 10" is in the command line as well - normally to kill a planner running in anytime mode. When you test from the terminal, do you use a timeout and if not does it take more than 10 seconds to find a solution? Perhaps removing this might solve your issue.

Best regards,
Michael

rakhmanu, 2021-05-27:
Hi again,

Thank you @m312z, yes that was the solution and it works perfectly. Now I am closing the issue.

closed 2021-05-27 by rakhmanu