KCL-Planning/ROSPlan #295: Problem in Domain and Problem

RodrigoCodeBernardo, 2021-06-17:
I am trying a domain and a planner with ROSPLAN, but I get errors on the domain and the problem.

I am using the popf planner. The domain and the problem I wrote them and tested using vscode. I tested using the popf planner.

Domain:

```
(define (domain UR3-Gripper)

  (:requirements :typing)
  (:types p_goal 
          obj 
          gripper)
  (:constants robotiq - gripper)
  
  (:predicates
    (at-robby ?r - p_goal)  ;; True iff x is a p_goal and the robot is in it
    (at ?b - obj ?r - p_goal)  ;; True if x is a obj, r a p_goal and x i in r
    (free ?g - gripper) ;; True if g is a gripper and does not hold a obj
    (carry ?o - obj ?g - gripper) ;; true if o is a obj, g a gripper and g holds o
    (open ?g - gripper)
  )

  ;; The robot can move from ?from to ?to
  (:action move_ur3
    :parameters (?from ?to - p_goal)
    :precondition (at-robby ?from)
    :effect (and (at-robby ?to)
                 (not (at-robby ?from))
                 )
  )

  ;; The robot can pick up ?obj in ?p_goal with ?gripper
  (:action pick
    :parameters (?obj - obj ?p_goal - p_goal ?gripper - gripper)
    :precondition (and (at ?obj ?p_goal) (at-robby ?p_goal) (free ?gripper) (open ?gripper))
    :effect (and (carry ?obj ?gripper ) (not (at ?obj ?p_goal))
            (not (free ?gripper))
            (not (open ?gripper))
            )
  )

  ;; Open Gripper
  (:action open
    :parameters (?gripper - gripper)
    :precondition (free ?gripper)
    :effect (open ?gripper)
  )

  ;; The robot can drop ?obj in ?p_goal from ?gripper
  (:action drop
    :parameters (?obj - obj ?p_goal - p_goal ?gripper - gripper)
    :precondition (and (carry ?obj ?gripper) (at-robby ?p_goal))
    :effect (and (at ?obj ?p_goal) (free ?gripper)
    (not (carry ?obj ?gripper))
    )
  )
)
```

Problem:

```
(define (problem UR3-Gripper)

  (:domain UR3-Gripper)

  (:objects
    p1 p2 - p_goal
    obj1 obj2 - obj
    robotiq - gripper)

  (:init
      (at-robby p1)
      (free robotiq)
      (at obj1 p1)
      (at obj2 p1))
 

  (:goal (and (at obj1 p2)
              (at obj2 p2)
              (at obj1 p2)))
  )
```

What can I be missing here?

m312z, 2021-06-17:
Could you try running with the planner you are using with ROSPlan, and if possible copy the error message here? Assuming that it does not generate a plan, then the error would be in the domain/problem somewhere.

If possible, copy and run the command line ROSPlan used to run the planner (it should appear as ROS INFO). That will run the planner on the automatically generated problem - perhaps there is an error introduced to the automatically generated problem file.

m312z, 2021-06-17:
One problem might be the use of constants:
(:constants robotiq - gripper)
The problem generator fetches all objects (including constants) which will duplicate domain constants in the problem file as objects. Some planners print a warning but continue to find a plan. Others might not.

This is fixed in one of the closed PRs #286 - the title of which is self-explanatory. This was merged Apr 14, so if that's the problem you might just need to do a git pull.

RodrigoCodeBernardo, 2021-06-17:
When I launch rosplan I immediately get this error:

```
[rosplan_knowledge_base-3] process has died [pid 18246, exit code -11, cmd /home/euroage/catkin_ws/devel/lib/rosplan_knowledge_base/knowledgeBase __name:=rosplan_knowledge_base __log:=/home/euroage/.ros/log/4283a316-cf78-11eb-b7ba-e0d55e204096/rosplan_knowledge_base-3.log].
log file: /home/euroage/.ros/log/4283a316-cf78-11eb-b7ba-e0d55e204096/rosplan_knowledge_base-3*.log
```
When I try to generate the plan these errors appear:

```
Critical Errors Encountered in Domain/Problem File
--------------------------------------------------

Due to critical errors in the supplied domain/problem file, the planner
has to terminate.  The errors encountered are as follows:
[ INFO] [1623942383.164066261]: KCL: (/rosplan_planner_interface) (problem.pddl) Planning complete
[ INFO] [1623942383.164318159]: KCL: (/rosplan_planner_interface) (problem.pddl) Plan was unsolvable.
[ INFO] [1623942384.223292870]: KCL: (/rosplan_parsing_interface) Parsing planner output.
[ERROR] [1623942384.228398523]: KCL: (/rosplan_parsing_interface) could not call Knowledge Base for predicate list
[ INFO] [1623942384.228995628]: KCL: (/rosplan_plan_dispatcher) Plan received.
[ INFO] [1623942385.374648309]: KCL: (/rosplan_plan_dispatcher) Dispatching plan.
[ INFO] [1623942385.574922999]: KCL: (/rosplan_plan_dispatcher) Dispatch complete.
[ERROR] [1623942385.580316696]: KCL: (/rosplan_plan_dispatcher) Failed to call service get_goals
```
I will then try removing the constants, the planner I used in vscode is the same one I am using in rosplan (popf).

RodrigoCodeBernardo, 2021-06-17:
Yes, the problem was really in the use of constants. I removed them and was able to generate the plan. I just find the error strange, because I use the same planner in rosplan and vscode.

Thanks for your help

```
rostopic echo /rosplan_planner_interface/planner_output -p
%time,field.data
1623942632333314895,0.000: (open robotiq)  [0.001]
0.001: (pick obj1 p1 robotiq)  [0.001]
0.002: (move_ur3 p1 p2)  [0.001]
0.003: (drop obj1 p2 robotiq)  [0.001]
0.004: (open robotiq)  [0.001]
0.004: (move_ur3 p2 p1)  [0.001]
0.005: (pick obj2 p1 robotiq)  [0.001]
0.006: (move_ur3 p1 p2)  [0.001]
0.007: (drop obj2 p2 robotiq)  [0.001]
```

closed 2021-06-17 by RodrigoCodeBernardo