KCL-Planning/ROSPlan #297: Problem with action_dispatch_topic

erfanshojaei, 2021-07-21:
Hello,

I am following "Tutorial 03 Plan Execution I":
https://kcl-planning.github.io/ROSPlan//tutorials/tutorial_03

Once, I enter "roslaunch tutorial_03.launch"  and then "rosrun rqt_graph rqt_graph", I don't see the topic of " /rosplan_plan_dispatcher/action_dispatch". 

Can anybody help me to solve this issue?

Thanks in advance,
Erfan

RodrigoCodeBernardo, 2021-07-22:
Hi,

Have you launched the action dispatch node?

	
```
	<node name="rosplan_plan_dispatcher" pkg="rosplan_planning_system" type="pddl_simple_plan_dispatcher" respawn="false" output="screen">
		<param name="knowledge_base"        value="rosplan_knowledge_base" />
		<param name="plan_topic"            value="/rosplan_parsing_interface/complete_plan" />
		<param name="action_dispatch_topic" value="action_dispatch" />
		<param name="action_feedback_topic" value="action_feedback" />
	</node>
```

Check your launch file.

erfanshojaei, 2021-07-22:
Hello,

Thank you for your message !

The problem still exists.

You may find my launch file and an image from rqt-graph of my program in 
the attached files.

Please let me know your idea.

If you need more information, please let me know.

Best Regards,

Erfan

-- 
=======================================================================================
Erfan Shojaei Barjuei, PhD. Postdoctoral Researcher in Robotics and Mechatronic Systems
Institut national des sciences appliquées de Rennes (INSA Rennes)
Département Génie Mécanique et Automatique (GMA)
Laboratoire de Génie Civil et Génie Mécanique (LGCGM) - EA 3913
20 avenue des Buttes de Coësmes
CS 70839 - 35 708 RENNES Cedex 7
FRANCE
E-mail: ***@***.***
Phone:  +33 2 23 23 89 36
Cell:   +33 6 65 45 28 39
=======================================================================================
Save a tree. Don't print this e-mail unless it's really necessary.

gerardcanal, 2021-07-26:
Hi Erfan,
As @RodrigoCodeBernardo was mentioning, it would have been good to see the topics listed when you run `rostopic list`. When running it, you should see the action_dispatch topic there. If you further run `rostopic info /rosplan_plan_dispatcher/action_dispatch`, you'll see that `/rosplan_plan_dispatcher` is the publisher, but no one is subscribed.

When I open rqt_graph I can't see the topic either. This is because the selector is on "Nodes only". This only shows the nodes that are active and the topics that link them together. As no one is yet subscribing to the action_dispatch topic, there is no link between `/rosplan_plan_dispatcher` node and any other nodes, so the topic is hidden.

If you select "Nodes/Topics (all)" in the first selector at the top-left of rt_graph, you should see all the topics involved, and you'll also see the action_dispatch topic is not yet connected to any node. This should change in the following tutorials when the simulated action interfaces are used.

I believe this should solve it, otherwise please feel free to reopen the issue.

closed 2021-07-26 by gerardcanal