KCL-Planning/ROSPlan #299: ROSPlan to Coppeliasim

erfanshojaei, 2021-07-31:
Hello,
My goal is to develop a plan to run a simulation of a manipulator in Coppeliasim. 
On this basis, I am following the tutorial_10. As you can see on tutorial_10, the two actions (undock , localise), are removed from the launch file.
If I add these two actions  Would you please let me know what I should do with Action Interface and CmakeLists.txt?   
How can I find the complete tutorial for my objective? 

Thanks and regards,
Erfan

gerardcanal, 2021-08-02:
Hi Erfan,
I have never worked with Coppeliasim so I can't help much on that front. In tutorial 10 it says to remove the simulated actions so it can teach the use of the abstract action interface to adapt the action interface to any actions or robots. The idea there is to generate the action interfaces for different kinds of actions, and having the robot doing something (as opposed to the simulated interfaces where there was no actual action being executed). 

Therefore, the tutorial removes the action interfaces to add new interfaces for the same actions that have an executor and can move the robot. If you follow the tutorial, all the necessary changes to the CMakelists and other files are detailed there.

erfanshojaei, 2021-08-02:
Hi Gerad,
Thank you for your support !
I have already followed your comments and solved my problem accordingly.
I have two questions. First) After launching file, there is no time during between actions. All the plan execute and complete in a few seconds !!!! :
-----------------------------------------------
[ INFO] [1627940624.194442715]: KCL: (/rosplan_planner_interface) (problem.pddl) Planning complete
[ INFO] [1627940624.194546370]: KCL: (/rosplan_planner_interface) (problem.pddl) Plan was solved.
[ INFO] [1627940624.194713091]: KCL: (/rosplan_parsing_interface) Planner output received.
[ INFO] [1627940625.121445129]: KCL: (/rosplan_parsing_interface) Parsing planner output.
[ INFO] [1627940625.277721924]: KCL: (/rosplan_plan_dispatcher) Plan received.
[ INFO] [1627940626.175121124]: KCL: (/rosplan_plan_dispatcher) Dispatching plan.
[ INFO] [1627940626.202075924]: KCL: (/rosplan_plan_dispatcher) Dispatching action [0, undock, 0.897191, 10.000000]
[ INFO] [1627940626.202588588]: KCL: (undock) action received
[ INFO] [1627940626.211798172]: KCL: (/rosplan_knowledge_base) Removing Fact (docked,0)
[ INFO] [1627940626.212086064]: KCL: (undock) TUTORIAL Action completing_erfan.
[ INFO] [1627940626.212157492]: KCL: (undock) action completed successfully
[ INFO] [1627940626.221779454]: KCL: (/rosplan_knowledge_base) Adding fact (undocked kenny, 0)
[ INFO] [1627940626.276071055]: KCL: (/rosplan_plan_dispatcher) Feedback received [0, action enabled]
[ INFO] [1627940626.276257048]: KCL: (/rosplan_plan_dispatcher) Feedback received [0, action achieved]
[ INFO] [1627940626.376030415]: KCL: (/rosplan_plan_dispatcher) 0: action undock completed
[ INFO] [1627940626.402193724]: KCL: (/rosplan_plan_dispatcher) Dispatching action [1, localise, 10.898192, 60.000000]
[ INFO] [1627940626.403479834]: KCL: (localise) action received
[ INFO] [1627940626.403734467]: KCL: (localise) TUTORIAL Action completing_erfan.
[ INFO] [1627940626.403850303]: KCL: (localise) action completed successfully
[ INFO] [1627940626.411899809]: KCL: (/rosplan_knowledge_base) Adding fact (localised kenny, 0)
[ INFO] [1627940626.475804215]: KCL: (/rosplan_plan_dispatcher) Feedback received [1, action enabled]
[ INFO] [1627940626.475911373]: KCL: (/rosplan_plan_dispatcher) Feedback received [1, action achieved]
[ INFO] [1627940626.576034883]: KCL: (/rosplan_plan_dispatcher) 1: action localise completed
[ INFO] [1627940626.602605279]: KCL: (/rosplan_plan_dispatcher) Dispatching action [2, goto_waypoint, 70.899190, 60.000000]
[ INFO] [1627940626.603818479]: KCL: (goto_waypoint) action received
[ INFO] [1627940626.611899038]: KCL: (/rosplan_knowledge_base) Removing Fact (robot_at,0)
[ INFO] [1627940626.612881155]: KCL: (goto_waypoint) TUTORIAL Action completing_erfan.
[ INFO] [1627940626.613015936]: KCL: (goto_waypoint) action completed successfully
[ INFO] [1627940626.622077259]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp1, 0)
[ INFO] [1627940626.622327956]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp1, 0)
[ INFO] [1627940626.676138359]: KCL: (/rosplan_plan_dispatcher) Feedback received [2, action enabled]
[ INFO] [1627940626.676332327]: KCL: (/rosplan_plan_dispatcher) Feedback received [2, action achieved]
[ INFO] [1627940626.776009468]: KCL: (/rosplan_plan_dispatcher) 2: action goto_waypoint completed
[ INFO] [1627940626.802631034]: KCL: (/rosplan_plan_dispatcher) Dispatching action [3, goto_waypoint, 130.900197, 60.000000]
[ INFO] [1627940626.804135995]: KCL: (goto_waypoint) action received
[ INFO] [1627940626.811987447]: KCL: (/rosplan_knowledge_base) Removing Fact (robot_at,0)
[ INFO] [1627940626.812900056]: KCL: (goto_waypoint) TUTORIAL Action completing_erfan.
[ INFO] [1627940626.813026913]: KCL: (goto_waypoint) action completed successfully
[ INFO] [1627940626.821928228]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp2, 0)
[ INFO] [1627940626.822088485]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp2, 0)
[ INFO] [1627940626.876210850]: KCL: (/rosplan_plan_dispatcher) Feedback received [3, action enabled]
[ INFO] [1627940626.876366056]: KCL: (/rosplan_plan_dispatcher) Feedback received [3, action achieved]
[ INFO] [1627940626.976084785]: KCL: (/rosplan_plan_dispatcher) 3: action goto_waypoint completed
[ INFO] [1627940627.002437375]: KCL: (/rosplan_plan_dispatcher) Dispatching action [4, goto_waypoint, 190.901189, 60.000000]
[ INFO] [1627940627.004122958]: KCL: (goto_waypoint) action received
[ INFO] [1627940627.011903262]: KCL: (/rosplan_knowledge_base) Removing Fact (robot_at,0)
[ INFO] [1627940627.012542366]: KCL: (goto_waypoint) TUTORIAL Action completing_erfan.
[ INFO] [1627940627.012769953]: KCL: (goto_waypoint) action completed successfully
[ INFO] [1627940627.021941687]: KCL: (/rosplan_knowledge_base) Adding fact (robot_at kenny wp3, 0)
[ INFO] [1627940627.022202166]: KCL: (/rosplan_knowledge_base) Adding fact (visited wp3, 0)
[ INFO] [1627940627.076202582]: KCL: (/rosplan_plan_dispatcher) Feedback received [4, action enabled]
[ INFO] [1627940627.076428914]: KCL: (/rosplan_plan_dispatcher) Feedback received [4, action achieved]
[ INFO] [1627940627.176178920]: KCL: (/rosplan_plan_dispatcher) 4: action goto_waypoint completed
[ INFO] [1627940627.202611137]: KCL: (/rosplan_plan_dispatcher) Dispatching action [5, goto_waypoint, 250.902196, 60.000000]
[ INFO] [1627940627.203852129]: KCL: (goto_waypoint) action received
[... cut ...]
[ INFO] [1627940639.891922313]: KCL: (/rosplan_knowledge_base) Removing Fact (dock_at,0)
[ INFO] [1627940649.891835819]: KCL: (/rosplan_knowledge_base) Adding fact (dock_at wp1, 0)
[ INFO] [1627940659.891865120]: KCL: (/rosplan_knowledge_base) Removing Fact (dock_at,0) my C++ file? for example which action is executing? where is the robot and what is the distention? what is the duration of the actions? if there is a function embedded in the action how I can extract the numeric values? ....

Thank you for your help in advance,
Erfan

erfanshojaei, 2021-08-02:
Hi Gerad,
Thank you for your support !
I have already followed your comments and solved my problem accordingly.
I have two questions. First) After launching file, there is no time during between actions. All the plan execute and complete in a few seconds !!!! 

My second question is: How I can extract the information of each action in my C++ file. 
for example which action is executing? where is the robot and what is the distention? what is the duration of the actions? if there is a function embedded in the action how I can extract the numeric values? ....

Thank you for your help in advance,
Erfan

gerardcanal, 2021-08-03:
Hi Erfan,
I'm not sure what is your first comment, but I fixed it to have the code part in code brackets (```) so it's easier to read.

If the plan executes at the same moment, this means that all the actions start at the same time. This means your domain allows for it to happen, and you should fix it. Basically, it means there are no preconditions that prevent those actions to execute. If you look at the plan string, you will see the actions start at the same time. 

For the second question, the tutorial shows how to create an action interface. The action interface callback will receive the dispatcher message, and this message contains all the information for the dispatched action (name and grounded parameters). The duration and dispatch time parameters are also included in that message (ActionDispatch). The action executing will be the one that receives the message from the dispatcher, but you can also subscribe to the action_dispatch topic to see which actions are being dispatched and when.

Regarding the rest, I am not sure what is a distention but where is the robot will depend on your implementation. You should be able to get the robot location from your move_base system and then you can try to find which is the closest waypoint to it. You can also look at the sensing interface where there are examples to achieve that. I am not sure what it means that there is a function embedded in the action, but all you need should be covered in the tutorials. If you need function values used on your domain, the Knowledge Base should have the updated values and you can query it as explained in the corresponding tutorials to get those values.

Best,
Gera**r**d

erfanshojaei, 2021-08-04:
Dear Gerard,
Thank you for your support !
Regarding the execution of actions in the same time, I am following the tutorials and use the materials (domain and problem) reported on the website. Moreover, I also tried  with my domain and problem files and see that there is no delay between actions. Since ROSPlan generates the plan as my expectations, I don't think so that there is problem in my domain and problem files.
You may find the files that I am currently using in the attached file, please let me know your opinion.
Thanks and regards,
Erfan

[Files_Erfan.zip](https://github.com/KCL-Planning/ROSPlan/files/6934872/Files_Erfan.zip)

gerardcanal, 2021-08-14:
Hi Erfan,
Apologies for my late reply as I was away and could not answer before. I assume you are using the simulated action interfaces for this. Can you show me the configuration of those in the launch file? I am thinking maybe the duration is not enough, and that is why the actions run so quickly.

erfanshojaei, 2021-08-15:
Hello,
Thank you for your message !
You may find my configuration files in the attachments.
If you need more information, please let me know.
Best Regards,
Erfan
[launch_files.zip](https://github.com/KCL-Planning/ROSPlan/files/6988935/launch_files.zip)

gerardcanal, 2021-08-16:
Hi Erfan,
You are not specifying action duration as a parameter, so this means the actions take the time specified in the plan. You can check the planner_output topic to see the duration of each action. This will be the time actions take to execute.

To specify action duration manually to a value X for each simulated action you can add the following parameter:
`		<param name="action_duration"		value="X" />`

Please try that in some actions to see if this changes the duration of the execution.

erfanshojaei, 2021-08-17:
Hello,
I added 
 <param name="action_duration" value="X" />
to each action in tutorial 10 (launch file)  but I don't see any changes in the execution time.
All actions run in the same time.
Regards,
Erfan

gerardcanal, 2021-08-18:
Can you show the esterel graph as shown in [tutorial 5](https://kcl-planning.github.io/ROSPlan//tutorials/tutorial_05)?

erfanshojaei, 2021-08-18:
Hello,
The esterel graph is as follows:
---------------------------------------------
rosmsg show rosplan_dispatch_msgs/EsterelPlan 

rosplan_dispatch_msgs/EsterelPlanNode[] nodes
  byte ACTION_START=0
  byte ACTION_END=1
  byte PLAN_START=2
  byte node_type
  int32 node_id
  string name
  rosplan_dispatch_msgs/ActionDispatch action
    int32 action_id
    int32 plan_id
    string name
    diagnostic_msgs/KeyValue[] parameters
      string key
      string value
    float32 duration
    float32 dispatch_time
  int32[] edges_out
  int32[] edges_in
rosplan_dispatch_msgs/EsterelPlanEdge[] edges
  byte CONDITION_EDGE=0
  byte START_END_ACTION_EDGE=1
  byte INTERFERENCE_EDGE=2
  byte edge_type
  int32 edge_id
  string edge_name
  int32 signal_type
  int32[] source_ids
  int32[] sink_ids
  float64 duration_lower_bound
  float64 duration_upper_bound
------------------------------------------------------------
if you need more information, please let me know.

Best Regards,
Erfan

gerardcanal, 2021-08-19:
Hi Erfan,
Please use the code option to display things like this as otherwise it is very hard to read.

This is not the esterel graph, this is the definition of the message. What I meant was to show the plot of the plan for the esterel dispatcher, as shown in the tutorials.

erfanshojaei, 2021-08-20:
Hello
Thank you for your support !
You may find the plan for the esterel dispatcher of mine and the tutorial in attached files. 
If you need more information, please let me know.
Regards,
Erfan

[plan.zip](https://github.com/KCL-Planning/ROSPlan/files/7024331/plan.zip)

erfanshojaei, 2021-08-20:
Is there any example for connecting ROSPlan to Gazebo?
Regards,
Erfan

dgerod, 2021-08-22:
_Is there any example for connecting ROSPlan to Gazebo?_

There is anything special to connect ROSPlan to Gazebo. The approach is the same as any connection to Gazebo from ROS, but you should call them inside ROSPlan actions instead of doing in a specific node. Therefore, you could check any ROS to Gazebo example, for example [this](https://github.com/ros-simulation/gazebo_ros_demos).

Let me do a brief explanation of how ROS to Gazebo connection is done. To connect them you should convert Gazebo topics and other of its mechanisms to ROS infrastructure. For doing that there are two approaches: (1) create a Gazebo plugin that convert Gazebo information to ROS, or (2) create a ROS node that reads Gazebo information using Gazebo infrastructure and converts it to ROS infrastructure: topics, services and actions. However, it could be possible that the plugins that you need are ready depending on what information you would like to use from Gazebo, see  this [repository](https://github.com/ros-simulation/gazebo_ros_pkgs).

My experience is with Gazebo Classic version, so I don't know if this approach could be follow in new ones.

erfanshojaei, 2021-08-31:
Hello,
Thank you for your support !

Is there any example of connecting the ROSPlan actions to ros moveit motion planner?

Thanks and regards,
Erfan

dgerod, 2021-09-04:
I don't know but it is not complex. In my case I use python so I do the connection using moveit_commander, see here an [example](http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html). For a C++ example you could see [here](https://github.com/ros-planning/moveit_tutorials/blob/master/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp).

Assuming that you are using C++, what you have to do is to implement a ROSPlan action of RPActionInterface and call MovetIt! inside the concreteCallback() method. And to know how to call MoveIt! you could check the tutorials above.