KCL-Planning/ROSPlan #302: FF Parsing Interface is parsing the cost as an action

silviaizquierdo, 2021-08-23:
Hello, 

When running the FFPlannerInterface on a plan generated by the Metric-FF planner, the parser is treating the cost as an action, which is then being dispatched. 

In my example, the plan generated can be seen below, along with the Esterel graph. 
`0: (move human wp6 wp4)  [0.001]`  
`1: (grasp human blue_box wp4)  [0.001]`
`2: (move human wp4 wp5)  [0.001]`
`3: (place human blue_box wp5)  [0.001]`
`4: (move human wp5 wp3)  [0.001]`
`5: (grasp human green_box wp3)  [0.001]`
`6: (move human wp3 wp2)  [0.001]`
`7: (place human green_box wp2)  [0.001]`
`plan (cost: 49.931942)  [0.001]`

[image]

After talking to @gerardcanal, I will modify line 108 in ROSPlan/rosplan_planning_system/src/PlannerInterface/FFPlannerInterface.cpp, so that the plan cost is not parsed as an action : `while (line.find("plan cost") == line.npos and line.find("Total cost") == line.npos and line.find("time spend") == line.npos and line.size() > 0)`

silviaizquierdo, 2021-08-23:
PR: [https://github.com/KCL-Planning/ROSPlan/pull/303](url)

gerardcanal, 2021-08-23:
Apparently, this was introduced by #234 in https://github.com/KCL-Planning/ROSPlan/pull/234/commits/af7ca5742948870df3ad8a6f717b36f94f8d5893#diff-2aaf07818c86a66c35bf6ec53cbac9cd856c1c52e9d0ef90767a0ddf04c2bc6bR108 although the previous version with "Cost of plan" should not work with the current Metric-FF version dispatched with ROSPlan.

PR #303 (2021-08-23): Update FFPlannerInterface
Updated FFPlannerInterface so that the plan cost is not parsed as an action. This fixes and closes #302.
closed 2021-08-23 by m312z