KCL-Planning/ROSPlan #305: Error when running Prost planner

yangshuo11, 2021-09-06:
Dear sirs, 
I am currently using the prost planner in ROSPlan framework. The "planner_command" is as: 
  <arg name="planner_command"  value=
    "$(find rosplan_planning_system)/common/bin/prost/run_prost_online.sh turtlebot_domain_bernoulli.rddl turtlebot_inst.rddl &quot;[PROST -s 1 -se [IPPC2014]]&quot;" />, and the dispatcher node selects the online plan dispatcher. When running the script "./turtlebot_explore_wp_from_file.bash", I get the following errors, and it seems that they come from the OnlinePlanDispatcher.cpp. Could you please help me with this issue? Thanks in advance.

[ INFO] [1630938506.832428227, 7.578000000]: KCL: (/rosplan_problem_interface) (problem.pddl) Generating problem file.
[ INFO] [1630938507.350563206, 7.800000000]: KCL: (/rosplan_problem_interface) (problem.pddl) The problem was generated.
[ INFO] [1630938507.352788928, 7.800000000]: KCL: (/rosplan_planner_interface) Problem received.
[ INFO] [1630938508.680801561, 8.374000000]: KCL: (/rosplan_planner_interface) (problem.pddl) Writing problem to file.
[ INFO] [1630938508.681215317, 8.374000000]: KCL: (/rosplan_planner_interface) (problem.pddl) Running: /home/ys/20210817-RAL-Project/RDDLPlanning/src/rosplan/rosplan_planning_system/common/bin/prost/run_prost_online.sh     /home/ys/20210817-RAL-Project/RDDLPlanning/src/rosplan_demos/rosplan_turtlebot3_demo/common/turtlebot_domain_bernoulli.rddl     /home/ys/20210817-RAL-Project/RDDLPlanning/src/rosplan_demos/rosplan_turtlebot3_demo/common/turtlebot_inst.rddl "[PROST -s 1 -se [THTS -act [UCB1] -out [UMC] -backup [PB] -init [Expand -h [IDS]] -t 1.0 ]]" > /home/ys/20210817-RAL-Project/RDDLPlanning/src/rosplan_demos/rosplan_turtlebot3_demo/common/plan.pddl in a new thread
[ INFO] [1630938508.683121733, 8.375000000]: KCL: (/rosplan_parsing_interface) Planner output received.
[ INFO] [1630938510.049798272, 8.945000000]: KCL: (/rosplan_parsing_interface) Parsing planner output.
[ INFO] [1630938511.384411496, 9.538000000]: KCL: (/rosplan_plan_dispatcher) Dispatching online plan
[ INFO] [1630938511.407538114, 9.553000000]: KCL: (/rosplan_plan_dispatcher) Starting IPPC server on port 3234 and waiting for connections!
[ INFO] [1630938512.519495943, 10.117000000]: KCL: (/rosplan_plan_dispatcher) Planner prost connected! Starting planning round
roundInitNode: 
<client>: ERROR: could not parse the clients's init session request.
Error writing to IPPC client! (-1).<client>: ERROR: no state response!! 
[ERROR] [1630938512.589137814, 10.153000000]: KCL: (/rosplan_plan_dispatcher) Could not obtain the planner response
Error writing to IPPC client! (-1).Error writing to IPPC client! (-1).[ INFO] [1630938513.571892570, 10.653000000]: KCL: (/rosplan_plan_dispatcher) Dispatch complete.
[ INFO] [1630938513.572046952, 10.653000000]: KCL: (/rosplan_plan_dispatcher) Total planning time: 0.000000 seconds

gerardcanal, 2021-09-06:
Hi @yangshuo11,
Happy to help there! Could you tell me which ROS version are you using (assuming melodic but we have not tested this much on noetic). Also, if you youse the sequential dispatcher does it work? (to rule out issues with the domain). Another thing to check is if the port number is correct in the online dispatcher node and the one running in prost (the planner command was not copied in your message).

yangshuo11, 2021-09-06:
Many thanks for your quick reply. 
My computer configuration is: ubuntu16.04 + ros kinetic. The planner_command is set to  "$(find rosplan_planning_system)/common/bin/prost/run_prost_online.sh turtlebot_domain_bernoulli.rddl turtlebot_inst.rddl &quot;[PROST -s 1 -se [IPPC2014]]&quot;". I have checked that all the port numbers are set to 3234 as default. 

This error occurs when I am testing the online planning & executing scheme of RDDL probabilistic planning in Tutorial 12. Following this tutorial, I have modified the "planner_command" as above, and set "planner_interface" to "online_planner_interface". The dispatcher_interface.launch is set as online_plan_dispatcher type according to Tutorial 12. The error occurs from the file "ippc_server.cpp" in rosplan_dependencies.

Previously, I have also tested the offline planning scheme as stated in Tutorial 12. In this test, I have set the "planner_command" as the same, and use the pddl_esterel_plan_dispatcher for plan dispatching. This scheme has been tested successfully.

Could you please provide any clues on the online planning test? I could found no relevant solutions around this issue anywhere on the internet. Thanks again.

gerardcanal, 2021-09-06:
Hi again,
I have been checking some things here and I am able to run the online dispatcher. Are you using the prost version that ships with ROSPlan?
If you could send me the files (launch + domain) I can try to reproduce the issue on my end. The online dispatcher works by emulating an IPPC server (which is what rddlsim does). It seems there is some issue with how prost sets the connection with the server, but I am not sure.

yangshuo11, 2021-09-07:
Thanks a lot. I am using the prost and rddlsim that embedded in ROSPlan. I have just tested these codes in ubuntu18.04+ros melodic, but failed again. I have attached my files and hope you would have any clues.
[online_prost_planning.tar.gz](https://github.com/KCL-Planning/ROSPlan/files/7119554/online_prost_planning.tar.gz)

gerardcanal, 2021-09-07:
Hi Yangshuo,
I managed to reproduce and also to run the online dispatcher with it. I found the issue, which is that the knowledge base and the problem generator were not set up correctly.

The first issue is that the problem path is not set up. While this is optional, setting it to the problem file you already have and use for planning is a good practice as it assures all the instances will be loaded to the KB (including non-fluents and so on). Otherwise, there must be someone providing the KB with all the information. This is in line 30 of turtlebot_main.launch.

The second one, is that the generated problem path is not set either (and it is set to a PDDL path). This means the problem generator will generate the problem in PDDL format instead of RDDL. While you overcomed this by forcing the domain and problem in the planner_command (more on this later), the OnlinePlannerInterface needs to know which is the problem path being used, and it gets this information from the KB. When asked the KB, it was receiving the following:
```
$ rosservice call /rosplan_planner_interface/get_planning_params 
domain_path: "/tmp/online_prost_planning/turtlebot_domain_bernoulli.rddl"
problem_path: "/home/gerard/code/ROSPlan_ws/src/rosplan/rosplan_planning_system/test/pddl/turtlebot/problem.pddl"
planner_ready: True
```
Thus, the Online dispatcher was sending this info to the IPPC server (i.e. "problem.pddl") and this one sends it back to the planner. Not sure about all the details of this interaction, but I believe the issue was this one.

This is now optional as I checked and it works with the fixed parameters, so with the changes above should be enough. The planner command should use the domain and problem paths obtained from the KB/ProblemGenerator.  The OnlineDispatcher keeps asking the KB for an updated state (and this state will have been updated from the action interfaces). Thus, to make sure the planner and the KB are speaking about the same state, it is recommended to use the DOMAIN and PROBLEM placeholders for the planning_command:

```
  <arg name="planner_command"  value=
    "$(find rosplan_planning_system)/common/bin/prost/run_prost_online.sh DOMAIN PROBLEM &quot;[PROST -s 1 -se [IPPC2014]]&quot;" />
```

This is specially important in the case of a replan, as otherwise the planner would plan from an initial state from the bginnning of the execution while the robot would be in a different one. Using the placeholders will make the planner use the generated problem file instead.

I think this should all fix you problem and the online dispatcher should now work. Let me know if that is the case.

yangshuo11, 2021-09-07:
Thanks for your warm help! I have managed to solve this issue. Best wishes to you! @gerardcanal

closed 2021-09-07 by yangshuo11