KCL-Planning/ROSPlan #311: KCL: (RPActionInterface) could not call Knowledge Base for operator details, goto_waypoint

karolinajudzentyte, 2022-03-23:
I have created my own action (very similar to rpmovebase) file. When trying to launch am getting:

[ INFO] [1648048101.336796716]: KCL: (/rosplan_knowledge_base) Ready to receive
[ INFO] [1648048101.340591621]: KCL: (/rosplan_planner_interface) Ready to receive
[ERROR] [1648048101.347032282]: KCL: (RPActionInterface) could not call Knowledge Base for operator details, goto_waypoint

Have no idea of how to go around to even debugging this. Any help would be appreciated

Launch file 
<launch>

    
    <arg name="map_file" default="$(find navigation_2d_spot)/maps/map.yaml"/>

    
    <include file="$(find navigation_2d_spot)/launch/navigate_no_move.launch" >
    </include>
    

    <node name="rosplan_roadmap_server" pkg="rosplan_interface_mapping" type="rpsimplemapServer" respawn="false" output="screen">
        <param name="data_path" value="$(find rosplan_planning_system)/common/" />
    </node>
 
    
    <rosparam command="load" file="$(find navigation_2d_spot)/config/waypoints.yaml"/>

    
    <include file="$(find rosplan_planning_system)/launch/interfaced_planning_system.launch" >
        <arg name="domain_path" value="$(find navigation_2d_spot)/common/no_move_base/domain_spot.pddl" />
        
        
        <arg name="problem_path" value="" />
        
        <arg name="autom_gen_problem_path" value="$(find navigation_2d_spot)/common/no_move_base/problem.pddl" />
        
        <arg name="data_path" value="$(find navigation_2d_spot)/common/no_move_base/" />
    </include>

    
    <node pkg="navigation_2d_spot" type="rpnavigate" name="RPNavigate" respawn="false" output="screen">
        <param name="knowledge_base" value="rosplan_knowledge_base" />
        <param name="pddl_action_name" value="goto_waypoint" />
        <param name="action_dispatch_topic" value="/rosplan_plan_dispatcher/action_dispatch" />
        <param name="action_feedback_topic" value="/rosplan_plan_dispatcher/action_feedback" />
    </node>

</launch>

domain: (define (domain spot_demo)

(:requirements :strips :typing :fluents :disjunctive-preconditions :durative-actions)

(:types
	waypoint 
	robot
)

(:predicates
	(robot_at ?v - robot ?wp - waypoint)
	(connected ?from ?to - waypoint)
	(visited ?wp - waypoint)
)

(:functions
	(distance ?wp1 ?wp2 - waypoint) 
)

;; Move to any waypoint, avoiding terrain
(:durative-action goto_waypoint
	:parameters (?v - robot ?from ?to - waypoint)
	:duration ( = ?duration 60)
	:condition (and
		(at start (robot_at ?v ?from))
	:effect (and
		(at end (visited ?to))
		(at end (robot_at ?v ?to))
		(at start (not (robot_at ?v ?from))))
)

karolinajudzentyte, 2022-03-24:
Hello, 

Thank you for such a quick reply. By running: 
rosservice call /rosplan_knowledge_base/domain/operator_details "name: 'goto_waypoint'" 
-> ERROR: service [/rosplan_knowledge_base/domain/operator_details] responded with an error: 
while running: 
rosservice call /rosplan_knowledge_base/domain/operators
-> operators: []

I can see that the goto_waypoint is not being added to the knowledge base. However, I don't know why.

Adding my header (same as used in move base)
#include <ros/ros.h>
#include <vector>
#include <iostream>
#include <fstream>
#include <actionlib/client/simple_action_client.h>
#include <navigate_2d_spot/NavigateToGoalAction.h>
#include <geometry_msgs/PoseStamped.h>
#include <std_srvs/Empty.h>
#include <tf2/LinearMath/Quaternion.h>
#include <rosplan_action_interface/RPActionInterface.h>

#ifndef KCL_movebase
#define KCL_movebase

namespace KCL_rosplan {

    class RPMoveBase: public RPActionInterface
    {

      public:
        RPMoveBase(std::string &actionserver);
        bool wpIDtoPoseStamped(std::string wpID, geometry_msgs::PoseStamped &result);
        bool concreteCallback(const rosplan_dispatch_msgs::ActionDispatch::ConstPtr& msg);

      private:

        actionlib::SimpleActionClient<navigate_2d_spot::NavigateToGoalAction> action_client_;
        ros::NodeHandle nh_;

        std::string waypoint_frameid_;
        std::string wp_namespace_;
    };
}
#endif

And my action server for ROSplan. Again very similar to move_bese (rpmovebase)

#include "include/RPNavigate.h"

namespace KCL_rosplan {

    // constructor
    RPNavigate::RPNavigate(std::string &actionserver) : action_client_(actionserver, true) {

        // create a node handle to manage communication with ROS network
        ros::NodeHandle nh("~");

        // get waypoints reference frame from param server
        nh.param<std::string>("waypoint_frameid", waypoint_frameid_, "map");
        nh.param<std::string>("wp_namespace", wp_namespace_, "/rosplan_demo_waypoints/wp");

        // setup a move base clear costmap client (to be able to send clear costmap requests later on)
        // clear_costmaps_client_ = nh.serviceClient<std_srvs::Empty>("/move_base/clear_costmaps");
        ROS_INFO("Constructor");
    }

    bool RPNavigate::wpIDtoPoseStamped(std::string wpID, geometry_msgs::PoseStamped &result) {
        ROS_INFO("Method");
[... 83 lines cut ...]
            }
        } else {
            // timed out (failed)
            action_client_.cancelAllGoals();
            ROS_INFO("KCL: (%s) action timed out", params.name.c_str());
            return false;
        }
    }
} // close namespace

int main(int argc, char **argv) {

    ros::init(argc, argv, "rosplan_interface_navigate");

    ros::NodeHandle nh("~");

    std::string actionserver;
    nh.param("action_server", actionserver, std::string("/navigate_to_goal_spot"));  // Python action node for navigation

    KCL_rosplan::RPNavigate rpmb(actionserver);

    rpmb.runActionInterface();

    return 0;
}

gerardcanal, 2022-03-24:
Hi,
I believe this is not related to the Action interface but in the loading of the domain in the KB. There's probably an issue in the domain and the KB can't load it. Can you check if there are any errors when the Knowledge Base launches?

gerardcanal, 2022-03-24:
Or actually, can you check if the domain is present in the path you specify?

`<arg name="domain_path" value="$(find navigation_2d_spot)/common/no_move_base/domain_spot.pddl" />`

You load the domain from there, so the file should be inside the navigation_2d_spot ros package, common/no_move_base/domain_spot.pddl. make sure the file is there.

You can check it by running the following in the terminal:
`cat $(rospack find navigation_2d_spot)/common/no_move_base/domain_spot.pddl`
Does this print your domain?

karolinajudzentyte, 2022-03-24:
Hello again,

Yes it does print:
cat $(rospack find navigation_2d_spot)/common/no_move_base/domain_spot.pddl
(define (domain spot_demo)

(:requirements :strips :typing :fluents :disjunctive-preconditions :durative-actions)

(:types
	waypoint 
	robot
)

(:predicates
	(robot_at ?v - robot ?wp - waypoint)
	(connected ?from ?to - waypoint)
	(visited ?wp - waypoint)
)

(:functions
	(distance ?wp1 ?wp2 - waypoint) 
)

;; Move to any waypoint, avoiding terrain
(:action goto_waypoint
	:parameters (?v - robot ?from ?to - waypoint)
	:condition (and
		(at start (robot_at ?v ?from))
	:effect (and
		(at end (visited ?to))
		(at end (robot_at ?v ?to))
		(at start (not (robot_at ?v ?from))))
)

)

and I can see with other service calls that predicates and functions are there

rosservice call /rosplan_knowledge_base/domain/functions
items: 
  - 
    name: "distance"
    typed_parameters: 
      - 
        key: "wp1"
        value: "waypoint"
      - 
        key: "wp2"
        value: "waypoint"

gerardcanal, 2022-03-24:
Hi again,
Your domain's parentheses are not right, that is why it doesn't load the actions properly. The parser for some reason can read it though.

You are missing a closing parenthesis at the end to close the domain, and another one after the at start condition to close the condition block (there should be 3 instead of 2).

This domain works:
```
(define (domain spot_demo)

(:requirements :strips :typing :fluents :disjunctive-preconditions :durative-actions)

(:types
waypoint
robot
)

(:predicates
(robot_at ?v - robot ?wp - waypoint)
(connected ?from ?to - waypoint)
(visited ?wp - waypoint)
)

(:functions
(distance ?wp1 ?wp2 - waypoint)
)

;; Move to any waypoint, avoiding terrain
(:durative-action goto_waypoint
:parameters (?v - robot ?from ?to - waypoint)
:duration ( = ?duration 60)
:condition (and
(at start (robot_at ?v ?from)))
:effect (and
(at end (visited ?to))
(at end (robot_at ?v ?to))
(at start (not (robot_at ?v ?from))))
)
)
```

closed 2022-03-24 by gerardcanal

karolinajudzentyte, 2022-03-24:
Thanks for your response. I did the change and I am still getting the same error. I had the same domain working with move base and then created an action interface for my own navigation, whit the same domain just a different launch action.

gerardcanal, 2022-03-24:
Hi Karolina,
Does the `rosservice call /rosplan_knowledge_base/domain/operator_details "name: 'goto_waypoint'"` still error? For me it works with the fixed domain.

karolinajudzentyte, 2022-03-24:
Hello Gerard,
Still getting the same error. Are you using my action interface?

karolinajudzentyte, 2022-03-24:
It works. Thank you for your help. It was a spelling error that was changed on the local file but not the robot.

gerardcanal, 2022-03-24:
Great! Glad to know it was fixed :-)