KCL-Planning/ROSPlan #316: fastDownward Planner selection in the ROSPlan launch file

bhomaidan1990, 2022-09-28:
I want to select the [FastDownward Planner](https://www.fast-downward.org) in [ROSPlan](https://github.com/KCL-Planning/ROSPlan) launch parameters.

ROSPlan has an [Interface to the FastDownward planner](https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/src/PlannerInterface/FDPlannerInterface.cpp), but the [official tutorial](https://kcl-planning.github.io/ROSPlan//documentation/interfaces/02_planner_interface.html) is not so obvious, and I'm not sure how to change the **planner command**: 
```xml
<arg name="planner_command" default="timeout 10 
$(find rosplan_planning_system)/common/bin/popf DOMAIN PROBLEM" />
```

##### Can you please tell me how can I select the FastDownward planner in the ROSPlan `launch` file parameters?
thanks in advance.

oscar-lima, 2022-09-28:
You need to use a bash command to call the planner as if you are calling it from the terminal.

bhomaidan1990, 2022-09-28:
@oscar-lima thanks for your rapid reply, if I understand your answer well you mean I have to call the **service** [planning_server_params](https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/src/PlannerInterface/FDPlannerInterface.cpp#L137) with the bash command as a parameter [req_planner_command](https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/src/PlannerInterface/PlannerInterface.cpp#L50)? is that what you mean or I missed your point? thanks in advance.

bhomaidan1990, 2022-10-04:
@oscar-lima I have tried to add the command as it is in the terminal:

 `python3 -m fast-downward.py  --alias lama-first domain.pddl problem.pddl` 

<details>
<summary>
To the Launch File (click to expand)
</summary>

```xml
<?xml version="1.0"?>
<launch>

	<!-- arguments -->
	<arg name="node_name"            default="rosplan_planner_interface" />
	<arg name="use_problem_topic"    default="true" />
	<arg name="problem_topic"        default="/rosplan_problem_interface/problem_instance" />
	<arg name="planner_topic"        default="planner_output" />
	<arg name="domain_path"	         default="$(find fd_planner_launch)/model/domain.pddl" />
	<arg name="problem_path"         default="$(find fd_planner_launch)/model/problem.pddl" />
	<arg name="data_path"            default="$(find fd_planner_launch)/model/" />
	<arg name="planner_command"      default="python -m $(find rosplan_planning_system)/common/bin/downward/fast-downward.py --alias lama-first DOMAIN PROBLEM" />

	<!-- knowledge base -->
	<node name="rosplan_knowledge_base" pkg="rosplan_knowledge_base" type="knowledgeBase" respawn="false" output="screen">
		<param name="domain_path" value="$(arg domain_path)" />
		<param name="problem_path" value="$(arg problem_path)" />
		<!-- conditional planning flags -->
		<param name="use_unknowns" value="false" />
	</node>
	
	<!-- plan parsing -->
	<node name="rosplan_parsing_interface" pkg="rosplan_planning_system" type="pddl_simple_plan_parser" respawn="false" output="screen">
		<param name="knowledge_base" value="rosplan_knowledge_base" />
		<param name="planner_topic"  value="/rosplan_planner_interface/planner_output" />
		<param name="plan_topic"     value="complete_plan" />
	</node>

	<!-- plan dispatching -->
	<node name="rosplan_plan_dispatcher" pkg="rosplan_planning_system" type="pddl_simple_plan_dispatcher" respawn="false" output="screen">
		<param name="knowledge_base"        value="rosplan_knowledge_base" />
		<param name="plan_topic"            value="/rosplan_parsing_interface/complete_plan" />
		<param name="action_dispatch_topic" value="action_dispatch" />
		<param name="action_feedback_topic" value="action_feedback" />
	</node>

	<!-- planner interface -->
	<node name="$(arg node_name)" pkg="rosplan_planning_system" type="fd_planner_interface" respawn="false" output="screen">

		<!-- read problem from topic (instead of from file) -->
		<param name="use_problem_topic" value="$(arg use_problem_topic)" />

		<!-- ros topics -->
		<param name="problem_topic" value="$(arg problem_topic)" />
		<param name="planner_topic" value="$(arg planner_topic)" />

		<!-- directory for files -->
		<param name="domain_path" value="$(arg domain_path)" />
		<param name="problem_path" value="$(arg problem_path)" />
		<param name="data_path" value="$(arg data_path)" />

		<!-- to run the planner -->
		<!-- planner command replaces strings 'DOMAIN' and 'PROBLEM' with paths -->
		<param name="planner_command" value="$(arg planner_command)" />
	</node>

</launch>

```
</details>

I have downloaded the `fastdownward` planner inside the ROSPlan `common/bin` folder as follows:
```sh
git clone https://github.com/aibasel/downward.git
cd downward
./build.py
```
and tested the planner with my simple working [domain](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/blob/main/fd_planner_launch/model/domain.pddl), and [problem](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/blob/main/fd_planner_launch/model/problem.pddl), here is my [planner launcher package](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/tree/main/fd_planner_launch), but still I can't run the fast downward planner, can you please explain to me how can I do that? thanks in advance.

oscar-lima, 2022-10-07:
Hi, you have to use the appropriate planner interface. Available planner interfaces can be consulted here: https://github.com/KCL-Planning/ROSPlan/tree/master/rosplan_planning_system/src/PlannerInterface . I assume the one you need is https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/src/PlannerInterface/FDPlannerInterface.cpp . Additionally by default ROSPlan uses popf planner interface, you have to change it. See:

planner_interface

https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/launch/interfaced_planning_system.launch#L25

https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_planning_system/CMakeLists.txt#L61

fd_planner_interface

closed 2022-10-07 by oscar-lima