KCL-Planning/ROSPlan #318: Calling problem_generation_server changes my problem's removing the disjunctive-preconditions

bhomaidan1990, 2022-10-17:
Hi,

I'm using the [modified ROSPlan](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/tree/main/ROSPlan) which works well with `ROS Noetic` using ubuntu `20.04` according to [this pull request](https://github.com/KCL-Planning/ROSPlan/pull/268).

I'm launching the [FastDownward Planning Interface](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/blob/main/fd_planner_launch/launch/rosplan_tutorial.launch),  I have downloaded the `fastdownward` planner inside the ROSPlan `common/bin` folder as follows:
```sh
git clone https://github.com/aibasel/downward.git
cd downward
./build.py
```
Whenever I call the `rosservice call /rosplan_problem_interface/problem_generation_server` it removes my `disjunctive-preconditions` causing [my pddl problem](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/blob/main/fd_planner_launch/model/problem.pddl) to be non-solvable!

<details>
<summary>
The Modified Problem After Calling the `problem_generation_server` service
</summary>

```
(define (problem task)
(:domain lego_simple)
(:objects
    rc1 rc2 rc3 rc4 - red_cube
    bc1 bc2 - blue_cube
    yc1 - yellow_cube
    lc1 lc2 - light_cube
    oc1 oc2 - olive_cube
    rb1 - red_brick
    bb1 bb2 - blue_brick
    yb1 yb2 yb3 - yellow_brick
    left_arm right_arm - robot
    operator - human
    p_02_09_0 p_06_13_0 p_05_13_3 p_02_10_4 p_05_14_0 p_05_13_2 p_02_10_3 p_05_12_2 p_05_12_1 p_02_11_3 p_02_10_2 p_02_11_2 p_05_13_1 p_05_14_1 p_03_11_0 p_04_11_0 p_02_10_0 p_02_11_0 p_02_10_1 p_02_11_1 p_05_12_0 p_05_13_0 platform - pos
)
(:init
    (in_stock rc1 left_arm)
    (in_stock rc2 operator)
    (in_stock rc3 operator)
    (in_stock rc4 operator)
    (in_stock yc1 operator)
    (in_stock bc1 left_arm)
    (in_stock bc2 operator)
    (in_stock lc1 operator)
    (in_stock lc2 operator)
    (in_stock oc1 left_arm)
    (in_stock oc2 operator)
    (in_stock yb1 left_arm)
    (in_stock yb2 operator)
    (in_stock yb3 operator)
    (in_stock bb1 left_arm)
    (in_stock bb2 operator)
    (in_stock rb1 operator)

    (empty p_02_09_0)
    (empty p_06_13_0)
    (empty p_05_13_3)
    (empty p_02_10_4)
    (empty p_05_14_0)
    (empty p_05_13_2)
    (empty p_02_10_3)
    (empty p_05_12_2)
    (empty p_05_12_1)
    (empty p_02_11_3)
    (empty p_02_10_2)
    (empty p_02_11_2)
    (empty p_05_13_1)
    (empty p_05_14_1)
    (empty p_03_11_0)
    (empty p_04_11_0)
    (empty p_02_10_0)
    (empty p_02_11_0)
    (empty p_02_10_1)
    (empty p_02_11_1)
    (empty p_05_12_0)
    (empty p_05_13_0)

    (under platform p_02_09_0)
    (under platform p_06_13_0)
    (under p_05_13_2 p_05_13_3)
    (under p_02_10_3 p_02_10_4)
    (under platform p_05_14_0)
    (under p_05_13_1 p_05_13_2)
    (under p_02_10_2 p_02_10_3)
    (under p_05_12_1 p_05_12_2)
    (under p_05_12_0 p_05_12_1)
    (under p_02_11_2 p_02_11_3)
    (under p_02_10_1 p_02_10_2)
    (under p_02_11_1 p_02_11_2)
    (under p_05_13_0 p_05_13_1)
    (under p_05_14_0 p_05_14_1)
    (under platform p_03_11_0)
    (under platform p_04_11_0)
    (under platform p_02_10_0)
    (under platform p_02_11_0)
    (under p_02_10_0 p_02_10_1)
    (under p_02_11_0 p_02_11_1)
    (under platform p_05_12_0)
    (under platform p_05_13_0)

)
(:goal (and
    (cube_at rc1 p_02_09_0)
    (cube_at rc2 p_06_13_0)
    (cube_at rc3 p_05_13_3)
    (cube_at rc4 p_02_10_4)
    (cube_at rc1 p_02_09_0)
    (cube_at rc2 p_06_13_0)
    (cube_at rc3 p_02_10_4)
    (cube_at rc4 p_05_13_3)
    (cube_at rc1 p_02_09_0)
    (cube_at rc2 p_05_13_3)
    (cube_at rc3 p_06_13_0)
    (cube_at rc4 p_02_10_4)
    (cube_at rc1 p_02_09_0)
    (cube_at rc2 p_05_13_3)
    (cube_at rc3 p_02_10_4)
    (cube_at rc4 p_06_13_0)
    (cube_at rc1 p_02_09_0)
    (cube_at rc2 p_02_10_4)
    (cube_at rc3 p_06_13_0)
    (cube_at rc4 p_05_13_3)
    (cube_at rc1 p_02_09_0)
    (cube_at rc2 p_02_10_4)
    (cube_at rc3 p_05_13_3)
    (cube_at rc4 p_06_13_0)
    (cube_at rc1 p_06_13_0)
    (cube_at rc2 p_02_09_0)
    (cube_at rc3 p_05_13_3)
    (cube_at rc4 p_02_10_4)
    (cube_at rc1 p_06_13_0)
    (cube_at rc2 p_02_09_0)
    (cube_at rc3 p_02_10_4)
    (cube_at rc4 p_05_13_3)
    (cube_at rc1 p_06_13_0)
    (cube_at rc2 p_05_13_3)
    (cube_at rc3 p_02_09_0)
    (cube_at rc4 p_02_10_4)
    (cube_at rc1 p_06_13_0)
    (cube_at rc2 p_05_13_3)
    (cube_at rc3 p_02_10_4)
    (cube_at rc4 p_02_09_0)
    (cube_at rc1 p_06_13_0)
    (cube_at rc2 p_02_10_4)
    (cube_at rc3 p_02_09_0)
    (cube_at rc4 p_05_13_3)
    (cube_at rc1 p_06_13_0)
    (cube_at rc2 p_02_10_4)
    (cube_at rc3 p_05_13_3)
    (cube_at rc4 p_02_09_0)
    (cube_at rc1 p_05_13_3)
    (cube_at rc2 p_02_09_0)
    (cube_at rc3 p_06_13_0)
    (cube_at rc4 p_02_10_4)
    (cube_at rc1 p_05_13_3)
    (cube_at rc2 p_02_09_0)
    (cube_at rc3 p_02_10_4)
    (cube_at rc4 p_06_13_0)
    (cube_at rc1 p_05_13_3)
    (cube_at rc2 p_06_13_0)
    (cube_at rc3 p_02_09_0)
    (cube_at rc4 p_02_10_4)
    (cube_at rc1 p_05_13_3)
    (cube_at rc2 p_06_13_0)
    (cube_at rc3 p_02_10_4)
    (cube_at rc4 p_02_09_0)
    (cube_at rc1 p_05_13_3)
    (cube_at rc2 p_02_10_4)
    (cube_at rc3 p_02_09_0)
    (cube_at rc4 p_06_13_0)
    (cube_at rc1 p_05_13_3)
    (cube_at rc2 p_02_10_4)
    (cube_at rc3 p_06_13_0)
    (cube_at rc4 p_02_09_0)
    (cube_at rc1 p_02_10_4)
    (cube_at rc2 p_02_09_0)
    (cube_at rc3 p_06_13_0)
    (cube_at rc4 p_05_13_3)
    (cube_at rc1 p_02_10_4)
    (cube_at rc2 p_02_09_0)
    (cube_at rc3 p_05_13_3)
    (cube_at rc4 p_06_13_0)
    (cube_at rc1 p_02_10_4)
    (cube_at rc2 p_06_13_0)
    (cube_at rc3 p_02_09_0)
    (cube_at rc4 p_05_13_3)
    (cube_at rc1 p_02_10_4)
    (cube_at rc2 p_06_13_0)
    (cube_at rc3 p_05_13_3)
    (cube_at rc4 p_02_09_0)
    (cube_at rc1 p_02_10_4)
    (cube_at rc2 p_05_13_3)
    (cube_at rc3 p_02_09_0)
    (cube_at rc4 p_06_13_0)
    (cube_at rc1 p_02_10_4)
    (cube_at rc2 p_05_13_3)
    (cube_at rc3 p_06_13_0)
    (cube_at rc4 p_02_09_0)
    (cube_at yc1 p_05_14_0)
    (cube_at bc1 p_05_13_2)
    (cube_at bc2 p_02_10_3)
    (cube_at bc1 p_02_10_3)
    (cube_at bc2 p_05_13_2)
    (cube_at lc1 p_05_12_2)
    (cube_at oc1 p_05_12_1)
    (cube_at oc2 p_02_11_3)
    (cube_at oc1 p_02_11_3)
    (cube_at oc2 p_05_12_1)
    (brick_at yb1 p_02_10_2 p_02_11_2)
    (brick_at yb2 p_05_13_1 p_05_14_1)
    (brick_at yb3 p_03_11_0 p_04_11_0)
    (brick_at yb1 p_02_10_2 p_02_11_2)
    (brick_at yb2 p_03_11_0 p_04_11_0)
    (brick_at yb3 p_05_13_1 p_05_14_1)
    (brick_at yb1 p_05_13_1 p_05_14_1)
    (brick_at yb2 p_02_10_2 p_02_11_2)
    (brick_at yb3 p_03_11_0 p_04_11_0)
    (brick_at yb1 p_05_13_1 p_05_14_1)
    (brick_at yb2 p_03_11_0 p_04_11_0)
    (brick_at yb3 p_02_10_2 p_02_11_2)
    (brick_at yb1 p_03_11_0 p_04_11_0)
    (brick_at yb2 p_02_10_2 p_02_11_2)
    (brick_at yb3 p_05_13_1 p_05_14_1)
    (brick_at yb1 p_03_11_0 p_04_11_0)
    (brick_at yb2 p_05_13_1 p_05_14_1)
    (brick_at yb3 p_02_10_2 p_02_11_2)
    (brick_at bb1 p_02_10_0 p_02_11_0)
    (brick_at bb2 p_02_10_1 p_02_11_1)
    (brick_at bb1 p_02_10_1 p_02_11_1)
    (brick_at bb2 p_02_10_0 p_02_11_0)
    (brick_at rb1 p_05_12_0 p_05_13_0)
))
)
```
</details>

I have tested [my domain](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/blob/main/fd_planner_launch/model/domain.pddl) and [my problem](https://gricad-gitlab.univ-grenoble-alpes.fr/pacbot_group/decision_making_pacbot/-/blob/main/fd_planner_launch/model/problem.pddl) with the [fast downward planner](https://www.fast-downward.org/) and they work well, there is a plan, but when I call the `problem_generation_server` service my problem changes causing me unsolvable issue! can you please tell me how can I solve that please? thanks in advance.

gerardcanal, 2022-10-17:
Hi @bhomaidan1990,
The PDDL Problem Generator doesn't seem to be generating disjunctive goals (see https://github.com/KCL-Planning/ROSPlan/blob/639037230073997d3ed3fea7391f1e1b3c6b1ffa/rosplan_planning_system/src/ProblemGeneration/PDDLProblemGenerator.cpp#L199)

But the main issue I see is that I believe the goals are not being parsed into ROS structures (can u try to call the service /rosplan_knowledge_base/state/goals and see if that shows the disjunctions?). If it does, extending the PDDLProblemGenerator is easy. If not, there's some other core changes that would need to be done, including on ROS msgs.

If that is the case, I would suggest to write over the PDDL-generated problem file from ROSPlan to write your goal there. ROSPlan will then pass that file to the planner and the problem should be solvable. I know it's not ideal, but I believe it should work.

bhomaidan1990, 2022-10-17:
Thanks for your rapid reply @gerardcanal I have tested `rosservice call /rosplan_knowledge_base/state/goals brick_at`, and here is the output:
<details>
<summary>
Click to see the output.
</summary>

```sh
attributes: 
  - 
    knowledge_type: 1
    initial_time: 
      secs: 0
      nsecs:         0
    is_negative: False
    instance_type: ''
    instance_name: ''
    attribute_name: "cube_at"
    values: 
      - 
        key: "c"
        value: "rc1"
      - 
        key: "p"
        value: "p_02_09_0"
    function_value: 0.0
    assign_op: 0
    optimization: ''
    expr: 
      tokens: []
    ineq: 
      comparison_type: 0
      LHS: 
        tokens: []
      RHS: 
        tokens: []
      grounded: False
  - 
    knowledge_type: 1
    initial_time: 
      secs: 0
      nsecs:         0
    is_negative: False
    instance_type: ''
    instance_name: ''
    attribute_name: "cube_at"
    values: 
      - 
        key: "c"
        value: "rc2"
      - 
        key: "p"
        value: "p_06_13_0"
    function_value: 0.0
    assign_op: 0
    optimization: ''
    expr: 
      tokens: []
    ineq: 
      comparison_type: 0
      LHS: 
        tokens: []
      RHS: 
        tokens: []
      grounded: False
  - 
    knowledge_type: 1
    initial_time: 
      secs: 0
      nsecs:         0
    is_negative: False
[... 3598 lines cut ...]
      - 
        key: "b"
        value: "rb1"
      - 
        key: "p1"
        value: "p_05_12_0"
      - 
        key: "p2"
        value: "p_05_13_0"
    function_value: 0.0
    assign_op: 0
    optimization: ''
    expr: 
      tokens: []
    ineq: 
      comparison_type: 0
      LHS: 
        tokens: []
      RHS: 
        tokens: []
      grounded: False
```
</details>

I don't know how "*to write over the PDDL-generated problem file from ROSPlan*" Are there further details on how to do that in the tutorial? otherwise can you please explain to me how to do that? thanks in advance.

gerardcanal, 2022-10-17:
Hello,
I can't really see between all the output, can you check if that is all your predicates individually? My guess is that it only shows the predicates and assumes they are all in a conjunctive clause, but I may be wrong. Without being familiar with the domain, I can't see it in such a long output.

About the workaround, there is no tutorial about that, you would have to program it yourself. What I meant is from your code, opening the file that rosplan generated, and writing there (programmatically) your goal function. This would be like completing the output of the problem_generator yourself, but leaving the part of the initial state which I understand is the part you're most interested about. If that's not the case, I would skip the problem generator and use your own problem generator or problem as needed.

bhomaidan1990, 2022-10-17:
@gerardcanal Even if they were all my predicates, the main problem is still **squeezing the goal state from AND of OR blocks to a single huge AND block without any OR**, which results in an unsolvable problem. regarding the "*workaround*" it seems to be a good idea, thanks for your kind suggestion.

gerardcanal, 2022-10-17:
I know, just wanted to check if there was something there I was not seeing, but that's what I was expecting.

Hope the workaround works!

closed 2022-10-17 by gerardcanal