KCL-Planning/ROSPlan #324: global name 'get_kb_attributes' is not defined

Ogusch, 2023-02-13:
Hello,

i am trying to create a simple sensing interface for my project.

Info: I have following predicates:
```   
(:predicates
    	(robot_attable ?roboter - type_robot ?table - type_table)
        (free)
        (not_free)
)
```
Within my porblem.pddl i created 8 objects from type type_table and 1 object from type type_robot. My sensing.yaml:
```  
functions:
    - $(find asap)/scripts/asap_sensing_interface.py
topics:
    robot_attable:
        - params:
              - robot
              - tablea1
          topic: /amcl_pose
          msg_type: geometry_msgs/PoseWithCovarianceStamped
              - robot
              - tablea2
          topic: /amcl_pose
          msg_type: geometry_msgs/PoseWithCovarianceStamped
...
```

Within my sensing interface script there is following code:
```
def robot_attable(msg, params):
    
    assert(msg.header.frame_id == "map")
    assert(len(params) == 2)
    return_value = []

    attributes = get_kb_attributes("robot_attable")
```

Running this always gets me this error:
```
[ERROR] [1676313353.672237, 111.819000]: bad callback: <bound method RosplanSensing.subs_callback of <__main__.RosplanSensing instance at 0x7f9fce3c0eb0>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 748, in _invoke_callback
    cb(msg, cb_args)
  File "/home/rico/rp_ws/src/rosplan/rosplan_sensing_interface/scripts/sensing_interface.py", line 303, in subs_callback
    result = eval(python_string, globals(), locals())
  File "<string>", line 1, in <module>
  File "/home/rico/rp_ws/src/asap/scripts/asap_sensing_interface.py", line 13, in robot_attable
    attributes = get_kb_attributes("robot_attable")
NameError: global name 'get_kb_attributes' is not defined
```

What am i doing wrong? Thank you in advance.

RodrigoCodeBernardo, 2023-02-13:
Hi,

Take a look at the example that is provided [sensing_interface](https://github.com/KCL-Planning/ROSPlan/blob/master/rosplan_sensing_interface/scripts/sensing_interface.py). And replicate the example.

Ogusch, 2023-02-13:
Hi,

in this example, a complete sensing interface is programmed, isn't it? I am actually trying to extend the existing sensing interface from ROSPlan using a Python-based YAML config file.

Ogusch, 2023-02-13:
I actually used also this example as a template. In line 31, the same command is called. 
```
attributes = get_kb_attribute("robot_at")
```
However, I do not understand what I am doing differently?

gerardcanal, 2023-02-13:
Hi Ogusch,
As you see in the example template, the name of the method is `get_kb_attribute`, while in your example in the issue post you are calling it get_kb_attribute**s** (in plural). I assume this is just a typo then!

See the tutorial for other examples: https://kcl-planning.github.io/ROSPlan//tutorials/tutorial_11

Closing this at it seems to be just a typo, please feel free to reopen it if it was not the typo and found it to be something else! :-)

closed 2023-02-13 by gerardcanal