KCL-Planning/ROSPlan #329: [rosplan_knowledge_base-1] process has died [pid 26564, exit code -11,

agj-neu, 2024-07-15:
When utilizing the Tutorial 01 Problem Generation component of the instructions, an error is encountered after executing the launch file. What steps should I take to troubleshoot this issue? Below is the content of the error message, along with my launch file.

agj@agj-VirtualBox:~/ROSPlan$ roslaunch launch/tutorial_02.launch
... logging to /home/agj/.ros/log/6544c612-4277-11ef-9bf5-080027ec2229/roslaunch-agj-VirtualBox-26545.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://agj-VirtualBox:35595/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosplan_knowledge_base/domain_path: /home/agj/ROSPlan...
 * /rosplan_knowledge_base/problem_path: /home/agj/ROSPlan...
 * /rosplan_knowledge_base/use_unknowns: False
 * /rosplan_planner_interface/data_path: /home/agj/ROSPlan...
 * /rosplan_planner_interface/domain_path: /home/agj/ROSPlan...
 * /rosplan_planner_interface/planner_command: timeout 10 /home/...
 * /rosplan_planner_interface/planner_topic: planner_output
 * /rosplan_planner_interface/problem_path: /home/agj/ROSPlan...
 * /rosplan_planner_interface/problem_topic: /rosplan_problem_...
 * /rosplan_planner_interface/use_problem_topic: True
 * /rosplan_problem_interface/domain_path: /home/agj/ROSPlan...
 * /rosplan_problem_interface/knowledge_base: rosplan_knowledge...
 * /rosplan_problem_interface/planning_language: PDDL
 * /rosplan_problem_interface/problem_path: /home/agj/ROSPlan...
 * /rosplan_problem_interface/problem_topic: problem_instance
 * /rosversion: 1.14.13

NODES
  /
    rosplan_knowledge_base (rosplan_knowledge_base/knowledgeBase)
    rosplan_planner_interface (rosplan_planning_system/popf_planner_interface)
    rosplan_problem_interface (rosplan_planning_system/problemInterface)

ROS_MASTER_URI=http://localhost:11311

process[rosplan_knowledge_base-1]: started with pid [26564]
process[rosplan_problem_interface-2]: started with pid [26565]
process[rosplan_planner_interface-3]: started with pid [26574]
[ INFO] [1721043525.403440055]: KCL: (/rosplan_knowledge_base) Starting a PDDL Knowledge Base
[ INFO] [1721043525.408728614]: KCL: (/rosplan_problem_interface) Ready to receive
[ INFO] [1721043525.419448860]: KCL: (/rosplan_knowledge_base) Parsing domain
[ INFO] [1721043525.419515547]: KCL: (/rosplan_knowledge_base) Parsing domain: /home/agj/ROSPlan/PlanningFile/domain.pddl.
[ INFO] [1721043525.420065294]: KCL: (/rosplan_knowledge_base) Parsing initial state
[ INFO] [1721043525.420098423]: KCL: (/rosplan_knowledge_base) Parsing Problem File: /home/agj/ROSPlan/PlanningFile/problem.pddl.
[ INFO] [1721043525.430124182]: KCL: (/rosplan_planner_interface) Ready to receive
[rosplan_knowledge_base-1] process has died [pid 26564, exit code -11, cmd /home/agj/ROSPlan/devel/lib/rosplan_knowledge_base/knowledgeBase __name:=rosplan_knowledge_base __log:=/home/agj/.ros/log/6544c612-4277-11ef-9bf5-080027ec2229/rosplan_knowledge_base-1.log].
log file: /home/agj/.ros/log/6544c612-4277-11ef-9bf5-080027ec2229/rosplan_knowledge_base-1*.log

agj-neu, 2024-07-15:
<?xml version="1.0"?>
<launch>

	
	<arg name="domain_path" default="/home/agj/ROSPlan/PlanningFile/domain.pddl" />
	
	<arg name="problem_path" default="/home/agj/ROSPlan/PlanningFile/problem.pddl" />

	
		<node name="rosplan_knowledge_base" pkg="rosplan_knowledge_base" type="knowledgeBase" respawn="false" output="screen">
		<param name="domain_path" value="$(arg domain_path)" />
		<param name="problem_path" value="$(arg problem_path)" />
		
		<param name="use_unknowns" value="false" />
	</node>

	<include file="$(find mongodb_store)/launch/mongodb_store.launch">
		<arg name="db_path" value="$(find rosplan_knowledge_base)/common/mongoDB/"/>
	</include>

	
	<include file="$(find rosplan_planning_system)/launch/includes/problem_interface.launch">
		<arg name="knowledge_base"   value="rosplan_knowledge_base" />
		<arg name="domain_path"      value="$(arg domain_path)" />
		<arg name="problem_path"     value="$(arg problem_path)" />

		<arg name="problem_topic"    value="problem_instance" />
	</include>
</launch>

JumpLee639, 2024-07-20:
我和你遇到了同样的问题，经过检查我发现我的问题是domain文件与problem文件不匹配造成的，具体地说例如problem文件中的动作在domain中未定义等，更换匹配后的文件便可正确运行。希望对你有所帮助。

gerardcanal, 2024-08-15:
Hello,
Apologies for the delay, I missed this issue. This seems to be a problem in your domain or problem file, that makes the parser fail in the knowledge base. Make sure the domain and problem work.

closed 2024-08-15 by gerardcanal

agj-neu, 2024-08-16:
Thanks for your answer. As a beginner, I am aware of this problem. My domain and problem files can be solved on some online planning websites, but the mismatch problem mentioned above occurs under rosplan. Could you please provide me with some correct domain and problem files? I want to complete the simple use of rosplan correctly first.

gerardcanal, 2024-08-16:
Hello,
Every planner has their own particularities and not all deal with PDDL in the same way and some may be more robust to errors in the input files. In ROSPlan, when the parser encounters an error, this usually makes the knowledge base die. This is not very helpful or user-friendly, and we plan to improve this in future versions.

I guess you're trying to use popf with ROSPlan, so it may be good to try to run the planner without rosplan to see if the planner can parse it (i.e., ./popf domain.pddl problem.pddl). The executable can be found in rosplan_planning_system/common.
While it is dying when parsing, the parser in popf should be similar to the one the knowledge base is using. Another option is to use VAL (https://github.com/KCL-Planning/VAL) which may also help to find errors in the domain. ROSPlan uses a version of the VAL parser.

Given the output logs above it seems that the domain is parsed well, so the issue seems to be in the problem file (the knowledge base seems to die while parsing it).

Hope this helps! Otherwise feel free to email me your domain and problem or attach it here and I'll try to see if I can find the error.

agj-neu, 2024-08-19:
Thank you for sharing your thoughts on how to solve the problem. Please understand my late reply to the message. I will attach the code here。
(define (domain move-room)
  (:requirements :strips :negative-preconditions :typing)
  
  (:types room gripper ball - object) 
  
  (:predicates
    (at-robby ?r - room)
    (path ?r1 - room ?r2 - room)
      (room ?from - room)
)
  
  (:action move
    :parameters (?from - room ?to - room)
    :precondition 
      (and 
        (room ?from) 
        (room ?to) 
        (at-robby ?from)
        (path ?from ?to)
      )
    :effect 
      (and 
        (at-robby ?to)
        (not (at-robby ?from))
      )
  )
)

(define (problem solve)
  (:domain move-room)
  (:requirements :strips :negative-preconditions :typing)
  

  (:objects 
      rooma roomb roomc roomd roome - room
  )
  
  (:init
      (room rooma)
      (room roomb) 
      (room roomc)
      (room roomd)
      (room roome)

      (path rooma roomb)
      (path roomb roomc)
      (path roomc roome)
      (at-robby rooma)
  )
  
  (:goal
      (and
          (at-robby roome)
      )
  )
)
Thank you again for your patience。

gerardcanal, 2024-08-19:
Hello,
I just tried these and I do get a plan with no errros with rosplan and popf:
```
0.000: (move rooma roomb)  [0.001]
0.001: (move roomb roomc)  [0.001]
0.002: (move roomc roome)  [0.001]
```